cesanta/mongoose-os #202: Unable to flash

pedroherig, 2017-03-07:
I am trying to flash the mongoose os via both the UI and the console but I am facing the following issue:

C:\MOS>mos flash
Loaded MOS/esp32 version 1.0 (20170307-164714/???)
Using port COM4
Opening COM4...
Connecting to ESP32 ROM, attempt 1 of 10...
  Connected
Running flasher @ 460800...
Error: failed to run flasher: failed to run flasher stub: failed to write params: failed to start mem write: command mismatch: 8 vs 5

Has anyone faced this issue or know how to solve it?

Thanks
Pedro Herig

rojer, 2017-03-07:
`mos flash -v 4 --logtostderr` please

pedroherig, 2017-03-08:
Sorry for not posting a more complete log for the first time, I didn't know about it.

Now I have been able to flash it, but I am stuck at the Wifi configuration section. My ESP32 looks unstable, sometimes it can flash sometimes it cannot. And at the wifi configuration it outputs the following log:

C:\MOS>mos wifi XXXX XXXX -v 4 --logtostderr
Using port COM4
I0308 11:23:11.259252    9600 util.go:17] Using port COM4
I0308 11:23:11.285269    9600 mgrpc.go:172] Connecting to COM4 over tSerial
I0308 11:23:11.289273    9600 serial.go:48] Opening COM4...
I0308 11:23:11.305309    9600 serial.go:58] COM4 opened: &{0x13606e00 752 {0 0} {0 0} 0x136f71e0 0x136f7200}, err: <nil>
Getting configuration...
I0308 11:23:11.305812    9600 mgrpc.go:229] Started recv loop, codec: [streamConnectionCodec to COM4]
I0308 11:23:11.306315    9600 util.go:17] Getting configuration...
I0308 11:23:11.311816    9600 mgrpc.go:283] created a request with id 1241379568978
I0308 11:23:11.312817    9600 serial.go:131]  ...sent frame delimiter.
I0308 11:23:11.509308    9600 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0308 11:23:11.515808    9600 serial.go:131]  ...sent frame delimiter.
I0308 11:23:11.709322    9600 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0308 11:23:11.716833    9600 serial.go:131]  ...sent frame delimiter.

(This 2 lines repeat a few dozens between 11:23:11 and 11:23:18, but I removed it to shrink the log at this post)

I0308 11:23:18.517822    9600 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0308 11:23:18.540413    9600 serial.go:131]  ...sent frame delimiter.
I0308 11:23:18.717989    9600 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0308 11:23:18.740596    9600 serial.go:155] closing serial COM4
I0308 11:23:18.918180    9600 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0308 11:23:19.023467    9600 main.go:192] Error: Device handshake timeout
/Users/lsm/src/cesanta.com/common/go/mgrpc/codec/serial.go:136: Device handshake timeout
/Users/lsm/src/cesanta.com/common/go/mgrpc/codec/stream.go:233:
/Users/lsm/src/cesanta.com/common/go/mgrpc/mgrpc.go:287:
/Users/lsm/src/cesanta.com/fw/defs/config/config_service.go:68:
/Users/lsm/src/cesanta.com/mos/dev/dev_conn.go:68:
/Users/lsm/src/cesanta.com/mos/config.go:77:
/Users/lsm/src/cesanta.com/mos/main.go:111:
Error: Device handshake timeout

pedroherig, 2017-03-08:
I turn the chip off and then on again to try to reflash it and the previous error occurred again, the logs are below:

C:\MOS>mos flash -v 4 --logtostderr
Loaded MOS/esp32 version 1.0 (20170307-164714/???)
Using port COM4
I0308 11:34:52.666666    9672 util.go:17] Using port COM4
Opening COM4...
I0308 11:34:52.696687    9672 report.go:13] Opening COM4...
Connecting to ESP32 ROM, attempt 1 of 10...
I0308 11:34:52.708699    9672 report.go:13] Connecting to ESP32 ROM, attempt 1 of 10...
I0308 11:34:53.159824    9672 rom_client.go:104] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0308 11:34:53.159824    9672 slip.go:87] => (44) 0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555
I0308 11:34:53.163833    9672 rom_client.go:104] => {cmd:Sync(8) csum:0 arg(36):"0707122055555555555555555555555555555555555555555555555555555555..."}
I0308 11:34:53.164834    9672 slip.go:87] => (44) 0008240000000000070712205555555555555555555555555555555555555555555555555555555555555555
I0308 11:34:53.200858    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.200858    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.205860    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.206864    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.210863    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.211865    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.211865    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.212864    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.213864    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.213864    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.214865    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.218868    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.220870    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.221869    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.222870    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.224873    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
  Connected
I0308 11:34:53.225871    9672 report.go:13]   Connected
Running flasher @ 460800...
I0308 11:34:53.229878    9672 report.go:13] Running flasher @ 460800...
I0308 11:34:53.230876    9672 rom_client.go:303] RunStub: params: 1 @ 0x40090000; code: 1956 @ 0x40090004, entry @ 400905a8; data: 32 @ 0x3ffd800c
I0308 11:34:53.232877    9672 rom_client.go:229] memWrite(0x40090000, 4, 0x00000000)
I0308 11:34:53.233878    9672 rom_client.go:104] => {cmd:MemWriteStart(5) csum:0 arg(16):"04000000010000000010000000000940"}
I0308 11:34:53.234879    9672 slip.go:87] => (24) 000510000000000004000000010000000010000000000940
I0308 11:34:53.234879    9672 slip.go:43] <= (12) 010804000707122000000000
I0308 11:34:53.235879    9672 rom_client.go:148] <= {cmd:Sync(8) value:538052359 ok:true lastError:0 body(4):"00000000"}
I0308 11:34:53.349504    9672 main.go:192] Error: failed to run flasher: failed to run flasher stub: failed to write params: failed to start mem write: command mismatch: 8 vs 5
/Users/lsm/src/cesanta.com/mos/flash/esp/rom_client.go:166: command mismatch: 8 vs 5
/Users/lsm/src/cesanta.com/mos/flash/esp/rom_client.go:237: failed to start mem write
/Users/lsm/src/cesanta.com/mos/flash/esp/rom_client.go:311: failed to write params
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher_client.go:86: failed to run flasher stub
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher_client.go:67:
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher.go:120: failed to run flasher
/Users/lsm/src/cesanta.com/mos/flash.go:130:
/Users/lsm/src/cesanta.com/mos/main.go:111:
Error: failed to run flasher: failed to run flasher stub: failed to write params: failed to start mem write: command mismatch: 8 vs 5

pedroherig, 2017-03-08:
Just adding new info: I also tried it in a new computer, in a Linux VM and using the ESP32 dev board and I still have the same issue.

cpq, 2017-03-09:
@pedroherig what esp32 board you're using?

pedroherig, 2017-03-09:
I am using a ESP_Module_Testboard and my own PCB

pedroherig, 2017-03-09:
Is there any ESP32 dev board that has already been tested with Mongoose OS? I can buy it and check if I still face the same issue.

cpq, 2017-03-14:
got it, thank you.

Does flashing work with the esptool ?

pedroherig, 2017-03-15:
Yes, it works with the esptool.

rojer, 2017-03-15:
@pedroherig weird - the device sends more sync responses than we expect.
i made a change to connection logic, can you download new `mos.exe` from the site and try again?
if it still doesn't work, please post the log here.

pedroherig, 2017-03-15:
Thank you for making the change. I am not with the boards right now, but I expect to test and send you an answer in the next hours.

pedroherig, 2017-03-20:
Sorry for not answering before. We identified an issue in our PCB project and we had to correct it before moving forward. 

The change solved our issue. Thank you very much.

rojer, 2017-03-20:
thanks, good to know!

closed 2017-03-20 by rojer