cesanta/mongoose-os #203: Bluetooth support for ESP32

hg-smart, 2017-03-08:
Did not find an example for bluetooth support on mongoose/esp32. Is Bluetooth support available / planned for ESP32 ?

cpq, 2017-03-08:
Not available yet. Planned, of course.
That said, native esp-idf SDK API is always available under MongooseOS, so if SDK has it, there is nothing stopping you to use it.

applecrusher, 2017-04-18:
I would like to put this idea out there.

It would be nice if after mongoose is installed the first time, we could use the bluetooth to (easily) change wifi information, so if the esp32 is encased or sealed, it's not a hassle to change this. This is what I would like to do with the bluetooth and I am sure I am not alone.

cpq, 2017-04-18:
That's in development already.
We might have something next week.

cpq, 2017-07-15:
Initial support is done, https://mongoose-os.com/blog/bluetooth-support-for-esp32/

closed 2017-07-15 by cpq