cesanta/mongoose-os #204: Device Configuration not saved

fonix232, 2017-03-09:
I've installed MOS about four days ago and have been playing around with it. 

However I noticed that no matter how many times I save changes made on the Device Configuration screen, they do not get applied to my device. It's like the config is not written to the file system, even though if set up with the initial setup wizard, it works.

For example I use (for now) the HiveMQ broker. But upon first reboot, it resets back to the default mosquitto test broker, which seems to be unavailable from my location (it times out).

cpq, 2017-03-09:
Just to clarify, what OS and what device you're using?

fonix232, 2017-03-09:
Oh sorry.

Host OS is Windows 10, device is a LoLin v3 NodeMCU.

KooT, 2017-03-09:
I hava the same problem with the same setup Win10 64, LoLin 3. 
I am adding my MQTT server during instalation, but then i see in logs:

mqtt_global_connect  MQTT connecting to test.mosquitto.org:1883

When I am changing it in configuration i see this:

[image]

cpq, 2017-03-14:
@alashkin   Please resolve

asmundstavdahl, 2017-03-21:
Same issue here with NodeMCU v3 and Ubuntu 16.04 as host OS.

I noticed that, when I clicked "Save configuration" in the web UI, the message "Configuration saved" appears twice.

deanmaniatis, 2017-03-21:
Same behaviour observed with latest MacOSX and two boards, Wemos D1 R2 and a generic ESP32 dev. board. For me it's noticeable when i perform wifi changes.

alashkin, 2017-03-23:
Fixed in d3f5e6ea64f4b926bc911a1f7f079b31db817be7
Note: even with latest mos tool you probably need to force reload configuration page (Ctrl+F5 or similar)

apassemard, 2017-04-04:
Hey this bug did happen to me yesterday so not sure if the fix has been incorporated in the latest MongooseOS yet. I'm on a Mac and was trying a NodeMCU board. My issue was that the MQTT settings were not save and reverted to hiveMQ broker every time. The workaround was to set the MQTT via command line instead of the Web UI and that worked.

cpq, 2017-04-04:
@apassemard Antony, you're right, reopening.

cpq, 2017-04-11:
That now works reliably,

closed 2017-04-11 by cpq