cesanta/mongoose-os #207: [enhancement] Multiple MQTT brokers

muhlpachr, 2017-03-10:
Use case: simultaneous cloud MQTT broker and local MQTT broker.
Local MQTT broker can serve local clients even in case of Internet connectivity outage.

MobiusHorizons, 2017-03-10:
(honest question) Is that better than federating from the local MQTT broker to the cloud broker?

muhlpachr, 2017-03-10:
No if you would like to "survive" with cloud broker updates/feeds in case of local broker failure.

MobiusHorizons, 2017-03-10:
fair enough.

cpq, 2017-04-03:
@muhlpachr is that what https://aws.amazon.com/greengrass/ is trying to do?

cpq, 2017-04-09:
I am not sure that this fallback functionality should be on the Mongoose OS side.

If there is a local MQTT broker already, it makes sense for Mongoose OS to be configured to use it. Local broker should either forward traffic to the remote one, or spool if there is no connection.

closed 2017-04-09 by cpq

muhlpachr, 2017-04-10:
How local MQTT broker can do anything in case it dies (e.g. hw failure) ?

cpq, 2017-04-10:
Fair question.

However, implementing fallback is a non-trivial task. How a failure is detected, at which point the switch happens, how to go back to the main target, should the device spool data locally if both targets fail and during the switch, etc etc.. Then, a fair question will arise - if done for MQTT, why not for HTTP? WebSocket? How authentication is handled? Security-wise, this is a huge potential for attacks.

So, the answer is: no fallbacks at this point in time.

muhlpachr, 2017-04-11:
Why one would like to detect failure at client side if amount of traffic is not problem ?

Why client can not afford to establish more simultaneous connections to MQTT brokers ?

Why not HTTP or WebSocket ? Because there is not Pub/Sub abstraction. MQTT PUBLISH is not dependent on MQTT broker (address of MQTT broker is not used) and same for SUBSCRIBE.

MQTT has shadow service client and whole shadow abstraction does not have any issues with multiple parallel MQTT broker connections. 

Authentication data is just per MQTT broker configuration as it is now.

muhlpachr, 2017-04-11:
AWS GreenGrass is for moving Lambda functions to on-premise compute resources (or to manage on-premise computer resources same way as Lambda functions).  Do you expect to implement availability cluster MQTT broker in Lambda functions ?

markterrill, 2020-01-13:
Hi, I have a simple use case for being able to configure multiple MQTT servers.
Primary MQTT is GCP for command/control/status

Another MQTT is local for Home Assistant, ie mosquitto running on a customer's raspi.

Want to be able to selectively publish a message to a specific MQTT resource. Optional argument. If no argument it uses the default primary.

rojer, 2020-01-13:
@markterrill interesting. please file as a new issue under https://github.com/mongoose-os-libs/mqtt

markterrill, 2020-01-14:
> @markterrill interesting. please file as a new issue under https://github.com/mongoose-os-libs/mqtt
Cheers @rojer, have done so: 
https://github.com/mongoose-os-libs/mqtt/issues/14