cesanta/mongoose-os #210: Any plan for supporting nRF5x?

nvl1109, 2017-03-16:
Hi,

Thank you for your effort on this awesome OS. Do you have any plan to support nRF5x devices? Thank you.

cpq, 2017-03-23:
the networking stack works on nRF, see  https://www.hackster.io/dfrank/mongoose-embedded-networking-library-on-nrf51-and-nrf52-547b15

both nRF51x and nRF52x

Could you elaborate on your project please?

nvl1109, 2017-04-01:
Thank you for the info. I just think the nRF devices are very interest and useful in IoT projects (actually I'm studying & make some hobby development with nRF on Zephyr RTOS), so it is very nice if mongoose-os supports nRF devices.

cpq, 2017-04-03:
Totally agree.

As soon as we get a commercial project on nRF platform, we'll port MongooseOS on it.

closed 2017-04-03 by cpq