cesanta/mongoose-os #212: Unable to download firmware (esp8266)

Aterocana, 2017-03-19:
Hello, I have an ESP8266 NodeMCU board. I install mos with the command `curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/sh`, connected the board to USB (I can see it at /dev/ttyUSB0) and then tried to flash the firmware:
`mos flash mos-esp8266 2> log`
The content of log is the following:
```
Fetching https://mongoose-os.com/downloads/mos-esp8266.zip...
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x14 pc=0xc2d35691]

runtime stack:
runtime.throw(0x8704cee, 0x2a)
	/usr/local/go/src/runtime/panic.go:566 +0x7f
runtime.sigpanic()
	/usr/local/go/src/runtime/sigpanic_unix.go:12 +0x59

goroutine 62 [syscall, locked to thread]:
runtime.cgocall(0x8553fc0, 0xd788ceb0, 0x0)
	/usr/local/go/src/runtime/cgocall.go:131 +0x125 fp=0xd788ce88 sp=0xd788ce6c
net._C2func_getaddrinfo(0xab5fd78, 0x0, 0xd79812a0, 0xd78a32a0, 0x0, 0x0, 0x0)
	??:0 +0x4b fp=0xd788ceb0 sp=0xd788ce88
net.cgoLookupIPCNAME(0xd7980ec0, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/cgo_unix.go:146 +0x355 fp=0xd788cf70 sp=0xd788ceb0
net.cgoIPLookup(0xd79593c0, 0xd7980ec0, 0xf)
	/usr/local/go/src/net/cgo_unix.go:198 +0x2d fp=0xd788cfd0 sp=0xd788cf70
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1612 +0x1 fp=0xd788cfd4 sp=0xd788cfd0
created by net.cgoLookupIP
	/usr/local/go/src/net/cgo_unix.go:208 +0x11e

goroutine 1 [select]:
net/http.(*Transport).getConn(0xd78040a0, 0xd7988580, 0x0, 0xd7958f00, 0x5, 0xd7980ec0, 0x13, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/transport.go:890 +0x99d
net/http.(*Transport).RoundTrip(0xd78040a0, 0xd78a8580, 0xd78040a0, 0x0, 0x0)
	/usr/local/go/src/net/http/transport.go:367 +0xa97
net/http.send(0xd78a8580, 0x89a76b0, 0xd78040a0, 0x0, 0x0, 0x0, 0x0, 0xd7988540, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:256 +0x521
net/http.(*Client).send(0x8ba0f30, 0xd78a8580, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:146 +0x127
net/http.(*Client).doFollowingRedirects(0x8ba0f30, 0xd78a8580, 0x87550f8, 0x31, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:528 +0x95f
net/http.(*Client).Get(0x8ba0f30, 0xd7958f00, 0x31, 0xd78a31c8, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:418 +0x92
net/http.Get(0xd7958f00, 0x31, 0xd77c3ce0, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:393 +0x3f
cesanta.com/mos/flash/common.NewZipFirmwareBundle(0xd7958f00, 0x31, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/flash/common/fw_bundle_zip.go:34 +0xde4
main.flash(0x89ab9e0, 0xd77ec018, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/flash.go:84 +0xc4
main.run(0xd789f830, 0x89ab9e0, 0xd77ec018, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/main.go:113 +0xad
main.main()
	/go/src/cesanta.com/mos/main.go:198 +0x68e

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1612 +0x1

goroutine 19 [select]:
cesanta.com/vendor/github.com/golang/glog.(*loggingT).flushDaemon(0x8ba1360)
	/go/src/cesanta.com/vendor/github.com/golang/glog/glog.go:927 +0x14b
created by cesanta.com/vendor/github.com/golang/glog.init.1
	/go/src/cesanta.com/vendor/github.com/golang/glog/glog.go:441 +0x262

goroutine 60 [select]:
net.lookupIPContext(0x89aba20, 0xd7959340, 0xd7980ec0, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/lookup.go:122 +0x72d
net.internetAddrList(0x89aba20, 0xd7959340, 0x86e5cac, 0x3, 0xd7980ec0, 0x13, 0x0, 0x0, 0x0, 0x0, ...)
	/usr/local/go/src/net/ipsock.go:241 +0x4f1
net.resolveAddrList(0x89aba20, 0xd7959340, 0x86e6339, 0x4, 0x86e5cac, 0x3, 0xd7980ec0, 0x13, 0x0, 0x0, ...)
	/usr/local/go/src/net/dial.go:179 +0x6cc
net.(*Dialer).DialContext(0xd77e41c0, 0x89aba20, 0xd7959340, 0x86e5cac, 0x3, 0xd7980ec0, 0x13, 0x0, 0x0, 0x0, ...)
	/usr/local/go/src/net/dial.go:329 +0x4d9
net.(*Dialer).DialContext-fm(0x89ab9e0, 0xd77ec018, 0x86e5cac, 0x3, 0xd7980ec0, 0x13, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/transport.go:43 +0x68
net/http.(*Transport).dial(0xd78040a0, 0x89ab9e0, 0xd77ec018, 0x86e5cac, 0x3, 0xd7980ec0, 0x13, 0x0, 0x0, 0x0, ...)
	/usr/local/go/src/net/http/transport.go:826 +0x70
net/http.(*Transport).dialConn(0xd78040a0, 0x89ab9e0, 0xd77ec018, 0x0, 0xd7958f00, 0x5, 0xd7980ec0, 0x13, 0x847f901, 0x0, ...)
	/usr/local/go/src/net/http/transport.go:967 +0x199a
net/http.(*Transport).getConn.func4(0xd78040a0, 0x89ab9e0, 0xd77ec018, 0xd7980ee0, 0xd7959100)
	/usr/local/go/src/net/http/transport.go:885 +0x48
created by net/http.(*Transport).getConn
	/usr/local/go/src/net/http/transport.go:887 +0x3a0

goroutine 61 [select]:
net.cgoLookupIP(0x89aba20, 0xd7959340, 0xd7980ec0, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0, 0xd77a4a00)
	/usr/local/go/src/net/cgo_unix.go:209 +0x346
net.lookupIP(0x89aba20, 0xd7959340, 0xd7980ec0, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/lookup_unix.go:70 +0x82
net.glob..func11(0x89aba20, 0xd7959340, 0x8754fd0, 0xd7980ec0, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/hook.go:19 +0x54
net.lookupIPContext.func1(0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/lookup.go:119 +0x6d
internal/singleflight.(*Group).doCall(0x8ba0c78, 0xd789f920, 0xd7980ec0, 0xf, 0xd7980f80)
	/usr/local/go/src/internal/singleflight/singleflight.go:93 +0x21
created by internal/singleflight.(*Group).DoChan
	/usr/local/go/src/internal/singleflight/singleflight.go:86 +0x31f
```

The command `mos console` works perfectly and the board works with other firmwares. Any idea? Thanks.

Aterocana, 2017-03-19:
Forgot to say I'm on Arch Linux.
> uname -a
Linux 4.10.2-1-ARCH #1 SMP PREEMPT Mon Mar 13 17:13:41 CET 2017 x86_64 GNU/Linux

Aterocana, 2017-03-20:
It seems the problem is when mos is trying to download anything automatically: `https://mongoose-os.com/downloads/mos-esp8266.zip` fails as well.
I'm able to flash the firmware if I download it manually with `wget` and then flash the file.
Still I'd like to solve the "download" problem.

Aterocana, 2017-03-20:
UPDATE: basically, as I thought, the issue is related with fetching remote resources. I tried it in a Kali Linux (Debian based) virtual machine and it worked. The issue must be with Archlinux distributions. Any idea on how make it work there?

cpq, 2017-03-23:
@alashkin please take that on.

@Aterocana could you try to debug the download please: `curl -v -i ....` - maybe there are some redirects or something under archlinux?

Aterocana, 2017-03-23:
sure:
`curl -v -i https://mongoose-os.com/downloads/mos-esp8266.zip > log.txt 2> errors.txt`

(attached both logs)
[log.txt](https://github.com/cesanta/mongoose-os/files/864678/log.txt)
[errors.txt](https://github.com/cesanta/mongoose-os/files/864679/errors.txt)

cpq, 2017-04-09:
maybe it fails to unzip to a local file?

`mos flash esp8266 --v 4 --logtostderr ` please

Aterocana, 2017-04-10:
`mos flash esp8266 --v 4 --logtostderr 2> logerr.txt`
[logerr.txt](https://github.com/cesanta/mongoose-os/files/910729/logerr.txt)

rojer, 2017-04-11:
@Aterocana i think i'm going to need to install arch to try. which version are you using?

Aterocana, 2017-04-11:
Vanilla archlinux https://www.archlinux.org/
```
> uname -a
Linux 4.10.8-1-ARCH #1 SMP PREEMPT Fri Mar 31 16:50:19 CEST 2017 x86_64 GNU/Linux
```
Do you need anything else? Thanks for the help.

cpq, 2017-06-01:
@Aterocana if that's still an issue for you, please try again.

`mos update` then `mos flash esp8266`

Please reopen if still fails.

closed 2017-06-01 by cpq

francescou, 2017-06-17:
same issue on arch linux version 4.11.3-1-ARCH

here is the output of `strace mos update`:

```
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
open("/usr/lib32/libdl.so.2", O_RDONLY|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0\3\0\1\0\0\0p\n\0\0004\0\0\0"..., 512) = 512
fstat64(3, {st_mode=S_IFREG|0755, st_size=13696, ...}) = 0
mmap2(NULL, 16448, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xc4f00000
mmap2(0xc4f03000, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0x2000) = 0xc4f03000
close(3)                                = 0
mprotect(0xc4f03000, 4096, PROT_READ)   = 0
mprotect(0xc45c2000, 8192, PROT_READ)   = 0
munmap(0xc538d000, 201670)              = 0
rt_sigprocmask(SIG_BLOCK, [HUP USR1 USR2 PIPE ALRM CHLD TSTP URG VTALRM PROF WINCH IO], [], 8) = 0
getpid()                                = 5625
getpid()                                = 5625
getpid()                                = 5625
open("/sys/fs/kdbus/0-system/bus", O_RDWR|O_NOCTTY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
socket(AF_UNIX, SOCK_STREAM|SOCK_CLOEXEC|SOCK_NONBLOCK, 0) = 3
getsockopt(3, SOL_SOCKET, SO_RCVBUF, [212992], [4]) = 0
setsockopt(3, SOL_SOCKET, SO_RCVBUFFORCE, [8388608], 4) = -1 EPERM (Operation not permitted)
setsockopt(3, SOL_SOCKET, SO_RCVBUF, [8388608], 4) = 0
getsockopt(3, SOL_SOCKET, SO_SNDBUF, [212992], [4]) = 0
setsockopt(3, SOL_SOCKET, SO_SNDBUFFORCE, [8388608], 4) = -1 EPERM (Operation not permitted)
setsockopt(3, SOL_SOCKET, SO_SNDBUF, [8388608], 4) = 0
connect(3, {sa_family=AF_UNIX, sun_path="/var/run/dbus/system_bus_socket"}, 33) = 0
getsockopt(3, SOL_SOCKET, SO_PEERCRED, {pid=1, uid=0, gid=0}, [12]) = 0
clock_gettime(CLOCK_MONOTONIC, {tv_sec=19159, tv_nsec=730092267}) = 0
fstat64(3, {st_mode=S_IFSOCK|0777, st_size=0, ...}) = 0
getsockopt(3, SOL_SOCKET, SO_ACCEPTCONN, [0], [4]) = 0
getsockname(3, {sa_family=AF_UNIX}, [128->2]) = 0
geteuid32()                             = 1000
sendmsg(3, {msg_name=NULL, msg_namelen=0, msg_iov=[{iov_base="\0AUTH EXTERNAL ", iov_len=15}, {iov_base="31303030", iov_len=8}, {iov_bas
e="\r\nNEGOTIATE_UNIX_FD\r\nBEGIN\r\n", iov_len=28}], msg_iovlen=3, msg_controllen=0, msg_flags=0}, MSG_DONTWAIT|MSG_NOSIGNAL) = 51
getpid()                                = 5625
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x14} ---
rt_sigreturn({mask=[HUP USR1 USR2 PIPE ALRM CHLD TSTP URG VTALRM PROF WINCH IO]}) = 0
write(2, "fatal error: ", 13fatal error: )           = 13
write(2, "unexpected signal during runtime"..., 42unexpected signal during runtime execution) = 42
```

theherk, 2017-07-19:
I'm having a very similar issue on Arch.

```
fw/examples/mjs_base❯ sudo drill mongoose.cloud                                                                         cesanta.com/git/release !
;; ->>HEADER<<- opcode: QUERY, rcode: NOERROR, id: 5224
;; flags: qr rd ra ; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 0
;; QUESTION SECTION:
;; mongoose.cloud.      IN      A

;; ANSWER SECTION:
mongoose.cloud. 300     IN      A       52.208.120.157

;; AUTHORITY SECTION:

;; ADDITIONAL SECTION:

;; Query time: 30 msec
;; SERVER: 10.128.128.128
;; WHEN: Wed Jul 19 12:32:42 2017
;; MSG SIZE  rcvd: 48
fw/examples/mjs_base❯ sudo mos build --arch esp32                                                                       cesanta.com/git/release !
Connecting to https://mongoose.cloud, user test
Uploading sources (162080 bytes)
Error: Post https://mongoose.cloud/api/fwbuild/1.9/build: dial tcp: lookup mongoose.cloud: no such host
```

rkakrik, 2017-08-27:
Same problem on arch-linux(

VishnuEaswaran, 2018-02-08:
The issue still persist. Any solutions as of yet ??

dimonomid, 2018-02-08:
Trying to figure where the problem is; could you please run `mos` with `GODEBUG` env var set to `netdns=cgo` and see if it still fails? Like this:
```
$ GODEBUG=netdns=cgo mos flash esp8266
```

Anshu6943, 2018-02-08:
if you are using Linux then try the commands with sudo
like sudo mos build --arch esp8266
It worked for me and perhaps will work for you too

dimonomid, 2018-02-08:
I don't think that `sudo` is related here; it may only be required if the port at which the device is attached (like `/dev/ttyUSB0`) is not user-writable, which should be fixed by adding appropriate udev rules, not by sudoing this and that. We did have problems with the resolver in the past; back then it was just fixed by forcing pure Go resolver during the build, but maybe for some reason it fails on Arch.

dimonomid, 2018-02-08:
If the `GODEBUG` thing doesn't do the trick, then the best bet for you would be to build mos tool from source: https://github.com/cesanta/mos-tool (see instructions there, they are for Ubuntu, but I think you'll figure how to tailor them for arch :) )

The right thing for us is to build proper arch packages of course, but no ETA for that at the moment.

dimonomid, 2018-02-09:
Fyi we've created `PKGBUILD` files for archlinux; they are not in AUR yet, so here are instructions: https://github.com/cesanta/mos-tool#installing-on-arch-linux