cesanta/mongoose-os #217: Unable to connect in WIFI network

pedroherig, 2017-03-23:
Dear, 

I am testing the mongoose os in an ESP32 dev board but I cannot connect to my wifi network. The network is working fine because I executed the same steps using an ESP8266 and everything went fine.

Could you please assist me?

C:\Users\Pedro\Downloads>mos wifi NAME PASS --v 4 --logtostderr
Using port COM4
I0323 11:35:16.178548    3252 util.go:17] Using port COM4
I0323 11:35:16.207567    3252 mgrpc.go:172] Connecting to COM4 over tSerial
I0323 11:35:16.211573    3252 serial.go:48] Opening COM4...
I0323 11:35:16.231623    3252 serial.go:58] COM4 opened: &{0x136a8ff0 776 {0 0} {0 0} 0x13788340 0x13788360}, err: <nil>
Getting configuration...
I0323 11:35:16.231623    3252 mgrpc.go:229] Started recv loop, codec: [streamConnectionCodec to COM4]
I0323 11:35:16.234611    3252 util.go:17] Getting configuration...
I0323 11:35:16.238113    3252 mgrpc.go:283] created a request with id 1278271884506
I0323 11:35:16.239113    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:16.435887    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:16.439889    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:16.636043    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:16.640552    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:16.836371    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:16.841377    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:17.036615    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:17.042104    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:17.236921    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:17.242418    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:17.436968    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:17.443540    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:17.636976    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:17.644209    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:17.837259    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:17.844268    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:18.037276    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:18.044388    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:18.237323    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:18.246192    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:18.437344    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:18.446380    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:18.637417    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:18.646453    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:18.837432    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:18.846575    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:19.037638    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:19.046785    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:19.237652    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:19.247062    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:19.437656    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:19.447281    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:19.637681    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:19.647695    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:19.837707    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:19.848867    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:20.037749    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:20.049816    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:20.238298    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:20.249870    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:20.438875    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:20.450448    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:20.638942    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:20.650556    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:20.838998    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:20.851116    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:21.039442    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:21.051626    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:21.239913    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:21.252419    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:21.440033    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:21.453716    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:21.640039    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:21.654559    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:21.840045    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:21.855109    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:22.040872    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:22.055995    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:22.241121    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:22.257595    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:22.441332    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:22.457876    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:22.641378    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:22.658048    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:22.841767    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:22.858232    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:23.041772    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:23.059011    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:23.241820    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:23.259015    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:23.442229    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:23.459752    3252 serial.go:135]  ...sent frame delimiter.
I0323 11:35:23.642280    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:23.660404    3252 serial.go:159] closing serial COM4
I0323 11:35:23.842288    3252 stream.go:201] 0 bytes read, err %!q(<nil>), eof? false buffer: []
I0323 11:35:23.953160    3252 main.go:199] Error: Device handshake timeout
/Users/lsm/src/cesanta.com/common/go/mgrpc/codec/serial.go:140: Device handshake timeout
/Users/lsm/src/cesanta.com/common/go/mgrpc/codec/stream.go:233:
/Users/lsm/src/cesanta.com/common/go/mgrpc/mgrpc.go:287:
/Users/lsm/src/cesanta.com/fw/defs/config/config_service.go:68:
/Users/lsm/src/cesanta.com/mos/dev/dev_conn.go:68:
/Users/lsm/src/cesanta.com/mos/config.go:77:
/Users/lsm/src/cesanta.com/mos/main.go:114:
Error: Device handshake timeout

Regards,
Pedro Herig

cpq, 2017-04-03:
What board are you using?

pedroherig, 2017-04-03:
I am using this SiP:
https://www.seeedstudio.com/ESP-32S-Wifi-Bluetooth-Combo-Module-p-2706.html?gclid=Cj0KEQjw5YfHBRDzjNnioYq3_swBEiQArj4pdIRtX5bEtmClsdWZPfb6SVsYvf_bNxtVVuGKuDe8iGgaAihw8P8HAQ
In this board:
https://electronics.semaf.at/ESP32s-ESP-WROOM-32-Adapter-to-254mm

pedroherig, 2017-04-03:
I also tried to test this board with mongoose in another network and it connects. But it cannot connect to this specific network

cpq, 2017-04-05:
does other hardware / software connect to that network ?

pedroherig, 2017-04-05:
Yes, including an ESP8266 running Mongoose OS

cpq, 2017-04-05:
i don't see how we can help you to debug this, sorry.
You probably need to resort to other solution, not mongoose os.

closed 2017-04-11 by cpq