cesanta/mongoose-os #220: mDNS doesn

kzyapkov, 2017-03-31:


cpq, 2017-04-09:
Fixed in https://github.com/cesanta/mongoose-os/commit/992a2b6a37afa431960895acc4e81ebec6501d42

@kzyapkov please verify

kzyapkov, 2017-04-12:
We've bumped to the latest revision of mongoose-os -- CC3200 builds fail with:

```
error #10056: symbol "mg_hash_md5_v" redefined: first defined in
   "/app/devices/dinrelay/.build/mongoose-os.a<mongoose.o>"; redefined in
   "/app/devices/dinrelay/.build/mongoose-os.a<cc3200_crypto.o>"
error #10056: symbol "mg_hash_sha1_v" redefined: first defined in
   "/app/devices/dinrelay/.build/mongoose-os.a<mongoose.o>"; redefined in
   "/app/devices/dinrelay/.build/mongoose-os.a<cc3200_crypto.o>"
error #10010: errors encountered during linking;
   "/app/devices/dinrelay/.build/dinrelay4.elf" not built
```

Looks like this happened along with the HTTP Auth refactoring, looks like an easy fix.

cpq, 2017-04-12:
yes, @rojer please take a look

rojer, 2017-04-12:
@kzyapkov make sure you have [this](https://github.com/cesanta/mongoose-os/blob/7a1e7e62feaab8de8c747d0eaf28e046a7a967ad/fw/platforms/cc3200/Makefile.build#L115) in your repo.

closed 2017-04-12 by rojer