cesanta/mongoose-os #221: Mongoose OS mJS Mac Address

applecrusher, 2017-03-31:
I was wondering if there was a way to get the mac address via mJS. 

I have the following,   

> let mac_address = Sys.ro_vars.mac_address;

I thought this would access the sys_ro_vars_schema.json file to get the MAC Address. I may have a complete misunderstanding of this file. However this prints as undefined. Is there a way to access it via mJS, or is there a ffi way of calling it?

cpq, 2017-04-17:
https://mongoose-os.com/snippets.html?j7c5SQiitQrq00KM

closed 2017-04-17 by cpq

suculent, 2017-08-20:
However closed this is, it does not work on ESP32 and i'm wondering about ESP8266 then. There's an example of FFI way but that either crashes or returns just a uint8 that needs to be translated to hex form somehow (i believe I've seen an example on that too...)

cpq, 2017-08-21:
https://mongoose-os.com/docs/quickstart/mos-tool-tips.html#get-mac-address-and-other-system-info-in-javascript

suculent, 2017-08-21:
Thanks a lot. However the RPC.LOCAL currently does not work (yet) on my ESP32 with MOS 1.14 and using let rpc_local = Cfg.get("device.id”); returns error (probably because it does not contain any protocol)

So far I was able to use last 6 characters from device-id and prepend “5ECF7F” / “5CCF7F" depending on device type, but that’s a hack.

cpq, 2017-08-21:
if you include rpc-common and rpc-loopback libraries, RPC.LOCAL should work.

suculent, 2017-08-21:
Great, it would be worth noting which libraries are required for which mJS include.
BTW, I'm still ending with `Unsupported connection scheme in RPC.LOCAL` error.