cesanta/mongoose-os #225: (ATECC508A) ATCA.SetConfig fails after UserExtra set

carldunham, 2017-04-04:
Not sure if this is really a bug or _caveat emptor_, but it seems that once the UserExtra byte is set to anything other than 0x00, it cannot be set again (enforced by the chip).

Because the ATCA.SetConfig RPC call sends the entire config, nothing after UserExtra will get set. This includes the KeyConfigs, sadly.

It might be helpful to break up SetConfig so that it can be called for individual sections. The `atcab_write_bytes_zone()` function seems to support this quite nicely.

In any case, dropping this here in case someone else runs into it.

rojer, 2017-04-05:
so, even setting to the same value does not work? bummer.
i'll do it, but may not get around to it for a while. in the mean time, PRs welcome.
device's `ATCA.ConfigSet` handler is [here](https://github.com/cesanta/mongoose-os/blob/a73a27047c356d713167c4abf224f4f86d00cf80/fw/src/mgos_atca_service.c#L49).

carldunham, 2017-04-06:
Yeah, I guess the code could check to see if it's the same and non-zero, and skip over it. I may do that, will save me 0.75US. :-)