cesanta/mongoose-os #226: (ATECC508A) Slots 8-15 get no ECC love

carldunham, 2017-04-04:
A few places in the code prevent using slots above 7 for ECC, yet the chip doesn't seem to have that restriction.

Using a chip configured with:
```yaml
- num: 11
  slotconfig:
    readkey: 7
    # privatekeyslotconfig:
    #   extsignenable: true
    #   intsignenable: true
    #   ecdhenable: true
    #   ecdhtonextslot: false
    nomac: false
    limiteduse: false
    encryptread: false
    issecret: true
    writekey: 0
    writeconfig: 10
  keyconfig:
    private: true
    pubinfo: true
    keytype: ECC
    lockable: true
    reqrandom: true
    reqauth: false
    authkey: 0
    intrusiondisable: false
    x509id: 0
```

Can do this:

```sh
% mos -X atca-get-pub-key 11 --dry-run=false
Using port /dev/cu.SLAB_USBtoUART

AECC508A rev 0x5000 S/N 0x0123eb0040385270ee, config is locked, data is locked

Slot 11, public key:

0xa1, 0x42, 0xe4, 0x8b, 0x3a, 0x97, 0x64, 0x5d, 0x02, 0x90, 0x44, 0xa7, 0xbc, 0x65, 0xdd, 0x28,
0x92, 0x79, 0x1f, 0xc4, 0x39, 0xcd, 0x92, 0x3c, 0x5e, 0x94, 0x23, 0x01, 0x15, 0x93, 0xea, 0xa9,
0xd9, 0x5a, 0xee, 0x93, 0x34, 0xba, 0x8d, 0x3c, 0x37, 0xad, 0xda, 0xdd, 0x39, 0xf5, 0xed, 0xb3,
0xa4, 0x44, 0x8c, 0xaa, 0xbc, 0x5a, 0x0b, 0x89, 0x4f, 0x79, 0x85, 0x98, 0x5a, 0xc5, 0xa8, 0xff

GetPubKey successful.
```

rojer, 2017-04-04:
i think you are right here. i looked closed in the datasheet and ECC private key is the _suggested_ use for slots 0-7. when implementing this, i misinterpreted it. i'll relax the check.

carldunham, 2017-04-04:
Cool. I'm also working on a PR that allows setting public ECC keys in slots 8-15 from a PEM. It may be rough (first time with go), but hope to have something this evening.

rojer, 2017-04-04:
that should be a simple matter of relaxing the constraint, setting keys from pem is already supported.

cpq, 2017-06-01:
Gents,
Is this still an issue? Can we close it ?

rojer, 2017-06-01:
yes, still an issue.