cesanta/mongoose-os #227: json_scanf() confused with multiple %Bs

carldunham, 2017-04-05:
The following simple app demonstrates the issue:

https://gist.github.com/carldunham/23d53832c620a2e04aefcc03db8f30b3

If you call it with various combinations of "fact1" and "fact2", the results will vary:
```sh
% mos --port ws://192.168.1.168/rpc call My.Test '{"fact1": true, "fact2": true, "num": 2}'
{
  "num": 3,
  "fact1": false,
  "fact2": true,
  "or": true,
  "and": false
}
```

console output:
```
mongoose_ev_handler  0x3ffaf4f4 HTTP connection from 192.168.1.151:56956
mg_rpc_add_channel   0x3ffaf3e8 '' WS_in
mg_rpc_ev_handler    0x3ffaf3e8 CHAN OPEN
mg_rpc_ev_handler    0x3ffaf3e8 GOT FRAME (95): {"src":"mos","id":1279333619515,"method":"My.Test","args":{"fact1":true,"fact2":true,"num":2}}

mg_rpc_parse_frame   1279333619515 'mos' '' 'My.Test'
mangle_params        args.p={"fact1":true,"fact2":true,"num":2}}
  * , args.len=35, ri->args_fmt={num: %d, fact1:%B, fact2:%B}
mg_rpc_send_frame    0x3ffaf3e8 SEND FRAME (130): {"id":1279333619515,"src":"esp32_020750","dst":"mos","result":{"num": 3, "fact1": false, "fact2": true, "or": true, "and": false}} -> 1
mg_rpc_ev_handler    0x3ffaf3e8 FRAME SENT (1)
```

Hope that helps!

carldunham, 2017-04-07:
So the issue is that frozen.c is expecting a pointer to an `int` for booleans, not `bool`. the comments in `frozen.h` are clear about that. Some places in the code are passing `*bool` and getting away with it, by luck of the way the compiler is organizing them on the stack/heap. Others are passing `*int` properly.

I have a PR in progress that will fix (or eliminate) some in `fw/src/mgos_atca_service.c`, but there are some in `fw/src/mgos_updater_rpc.c` (and possibly elsewhere) that I would rather not touch right now.

ruslanvaliullin, 2017-04-17:
Added support for char-sized bools to frozen. Please check.

carldunham, 2017-04-17:
That should work, but what about existing code that uses *int?

Also, wouldn't it just be simpler to require *bool and let the compiler deal with the sizes?

ruslanvaliullin, 2017-04-18:
Thanks for question! Yes, but we do not want to change the API yet. We will do it in one of the future releases. Please make sure that your example works. Regards.

closed 2017-04-18 by ruslanvaliullin

carldunham, 2017-04-18:
So the calls I have that use int* are now broken, but changing them to use bool* works.