cesanta/mongoose-os #228: [enhancement] ESP32 Ethernet support

muhlpachr, 2017-04-06:
Reference HW https://olimex.wordpress.com/2017/02/09/esp32-evb-our-new-board-for-iot-first-prototypes-are-ready/

cpq, 2017-04-06:
Fantastic. We'll start on it shortly.

cpq, 2017-06-01:
Work in progress, cannot give an estimate yet, soon we'll have it.

tusharvb19, 2017-07-11:
It will be great if it works with LAN8720 and also with Wiznet5100 or ENC28J60.
As in IOT ehternet can be the main source for industrial gateways.
Waiting for the development and if there is any way we can use this hardware as we use in arduino using arduino API.
TIA.

rojer, 2017-07-19:
LAN87x0 and TLK100 are now supported, please see https://github.com/mongoose-os-libs/ethernet

ESP32 includes EMAC and only needs a PHY. I don't see a compelling reason to use ENC28J60 (which includes a MAC), let alone Wiznet 5100 (which is everything + a kitchen sink kinda chip).
May save a couple pins (though SPI is at least 3 pins, 4 with CS), but requires more complexity on the software side, for what seems like little benefit. @tusharvb19 if you see reasons for us to support these more full-featured controllers, please file a separate issue.

closed 2017-07-19 by rojer

tusharvb19, 2017-07-20:
Thanks for the work👍