cesanta/mongoose-os #229: Add auth hook for `POST /`

kzyapkov, 2017-04-07:


carldunham, 2017-04-07:
Or better yet, cryptographically signed

rojer, 2017-04-10:
as a first step, i am going to add a global password-based protection (using http digest auth) for all HTTP requests. static files, handlers - everything. this should take care of basic HTTP server protection, and should finer-grained control be required, we can consider it later.

rojer, 2017-04-11:
https://github.com/cesanta/mongoose-os/commit/2b26fa0be901f3365e7ad276ab196c40c4475a1a

with this, you should be able to generate a password file with the standard `htdigest` utility like so: `htdigest -c fs/htpasswd mydomain admin`, put it on the device's filesystem, configure `http.auth_domain=mydomain` and `http.auth_file=htpasswd`. with that, all HTTP accesses will require user `admin` with a valid password. `curl` supports digest auth, enable it with `--digest` in addition to the `-u admin:password` argument.

closed 2017-04-11 by rojer

kzyapkov, 2017-04-12:
We do require finer grained control -- some resources should be publicly available. Maybe make this controllable when registering the endpoint? A global auth setting doesn't really work for us.

kzyapkov, 2017-04-12:
Also, curiously, all resources *except* for `POST /update` require a password.

iliapenev, 2017-04-18:
mgos_register_http_endpoint_opt does not work on "/". It is still protected. 
Could you please fix it. Thanks!

rojer, 2017-04-18:
@iliapenev no, it works fine for me, i just tested. if you register your endpoint like this:

```c
  struct mg_http_endpoint_opts opts;
  memset(&opts, 0, sizeof(opts));
  mgos_register_http_endpoint_opt("/", hh, opts);
```

it handles all requests not handled by other endpoints, without authorization.

iliapenev, 2017-04-18:
I've tried with example from Mongoose-os - works fine. 
With our project does not. I will check again project and if there is still problems will let you know. 
Thanks!