cesanta/mongoose-os #231: MQTT drop

williamwd, 2017-04-11:
Using the c_mqtt example I got my ESP8266 (nodemcu) working but every 10 seconds the mqtt connection drops and it reconnects. The first time I solved in AWS' IoT panel (no policy attached to certificate) but now it didn't work. How do I debug this connection?

Thanks

cpq, 2017-04-12:
there could be multiple reasons. increase debug level, `mos config-set debug.level=3`

I suggest using `mos aws-iot-setup` first, to get a working baseline.

closed 2017-04-12 by cpq

williamwd, 2017-04-19:
I'll post here because I spent a week trying to find out the reason that aws would disconnect me without clearly saying the reason. 
AWS logs wouldn't tell me; Mongoose logs just showed a disconnect.
The problem happened because I had multiple ESP8266 connected to AWS IoT, so the `mqtt.client_id` was the same for both devices.
[Documentation](http://docs.aws.amazon.com/iot/latest/developerguide/protocols.html) says: 

> The message broker uses the client ID to identify each client. The client ID is passed in from the client to the message broker as part of the MQTT payload. **Two clients with the same client ID are not allowed to be connected concurrently to the message broker**. When a client connects to the message broker using a client ID that another client is using, a CONNACK message will be sent to both clients and the currently connected client will be disconnected.

So just set different client_id for each device: `mos config-set mqtt.client_id=device123`

Thanks!

cpq, 2017-04-20:
@williamwd thanks for letting us know, appreciated!