cesanta/mongoose-os #232: Build local with docker toolbox

nvl1109, 2017-04-12:
Hi, I'm using docker toolbox (virtualbox based) latest version on win10 64bit.

D drive was shared in virtualbox as "D" mount -> in docker I can access my D drive by "/D/<contents>" same as msys environment (which used by Docker Toolbox Terminal).
Mongoose-os was cloned into "/D/Works/esp32/code/mongoose-os" directory.
I pulled docker image docker.cesanta.com/esp32-build:1.0-r13 to my machine.

After that I use mos (latest version) to initial esp32 project
``mos init --arch esp32``
Then build locally
``mos build --verbose --logtostderr --local --repo /D/Works/esp32/code/mongoose-os``
And here are result:
```
$ mos build --verbose --logtostderr --local --repo /D/Works/esp32/code/mongoose-os
Using mongoose-os located at "D:/Works/esp32/code/mongoose-os"
Building...
Make arguments: -j -f D:\Works\esp32\code\mongoose-os\fw\platforms\esp32\Makefile FS_STAGING_DIR=build\fs BUILD_DIR=build\objs GEN_DIR=build\gen APP_FS_FILES=fs\* APP_CONF_SCHEMA=src/conf_schema.yaml APP_SOURCES=src\*.c src\*.cpp FFI_SYMBOLS= MGOS_PATH=D:/Works/esp32/code/mongoose-os FW_DIR=build\fw APP=tmp PLATFORM=esp32 APP_VERSION=1.0
D:/Works/esp32/code/mongoose-os/fw/docker_common.mk:90: warning: undefined variable `abspath D:/Works/esp32/code/mongoose-os'
D:/Works/esp32/code/mongoose-os/fw/docker_common.mk:90: warning: undefined variable `abspath D:/Works/esp32/code/mongoose-os'
D:/Works/esp32/code/mongoose-os/fw/docker_common.mk:90: warning: undefined variable `abspath D:/Works/esp32/code/mongoose-os'
Using docker.cesanta.com/esp32-build:1.0-r13
time="2017-04-12T16:33:10+07:00" level=info msg="Unable to use system certificate pool: crypto/x509: system root pool is not available on Windows"
D:\Setups\DockerToolbox\docker.exe: Error response from daemon: invalid volume spec ":/mongoose-os": invalid volume specification: ':/mongoose-os'.
See 'D:\Setups\DockerToolbox\docker.exe run --help'.
make: *** [all] Error 125
I0412 16:33:10.521564    3148 main.go:207] Error: exit status 2
exit status 2
/Users/lsm/src/cesanta.com/mos/build.go:326:
/Users/lsm/src/cesanta.com/mos/build.go:258:
/Users/lsm/src/cesanta.com/mos/build.go:64:
/Users/lsm/src/cesanta.com/mos/main.go:120:
Error: exit status 2

nvl11@LINH-LT MINGW64 /d/Works/esp32/tmp
$```

I just think this error seems because mongoose-os is not compatible with MSYS environment.
Do you have any idea to help me build mongoose-os app with my current environment? Thank you.

nvl1109, 2017-04-12:
I tried to run docker with mongoose-os mount, it seems ok.
```
$ docker run -v /D/Works/esp32/code/mongoose-os:/mongoose-os -ti docker.cesanta.com/esp32-build:1.0-r13
time="2017-04-12T16:49:50+07:00" level=info msg="Unable to use system certificate pool: crypto/x509: system root pool is not available on Windows"
root@991b2a4da0e6:/# ls /mongoose-os/
LICENSE  README.md  common  docs  frozen  fw  mongoose  mos  third_party  tools  vendor
```

nvl1109, 2017-04-12:
I guest the mos was translated the repo path from "/D/Works/esp32/code/mongoose-os" to "D:\Works\esp32\code\mongoose-os" in windows style, then mount docker volume with argument "-v D:\Works\esp32\code\mongoose-os:/mongoose-os". This could cause invalid volume spec ":/mongoose-os" failure as above.
Any idea? Thanks.

dimonomid, 2017-08-01:
Support for local builds in both "docker toolbox" and "docker for windows" was added in 1.12.

closed 2017-08-01 by dimonomid