cesanta/mongoose-os #233: Update esp-idf to v2.0

JonnyBGod, 2017-04-17:
Any planed timeframe for this?

cpq, 2017-04-18:
Sure, we update.
@JonnyBGod what's your particular interest in updating, is there some specific functionality/fix you're looking for?

JonnyBGod, 2017-04-18:
Need some of the fixes in Wifi and BT, in particular the promiscuous mode fix.

JonnyBGod, 2017-04-18:
Are there instructions on how to compile with a new version of esp-idf?

rojer, 2017-04-18:
@JonnyBGod we follow esp-idf pretty closely, the version we have in the build image at the moment is just a few days old - https://github.com/espressif/esp-idf/commit/3cad00fdcca7dd4b7939d7862407c000c29657c1

i tried to pull even more recent fixes, but in fact, ESP-IDF is currently broken: https://github.com/espressif/esp-idf/issues/520

i'm waiting for that bug to be fixed.

closed 2017-04-26 by rojer