cesanta/mongoose-os #234: Unable to set or get any config data with UART esp8266 firmware

bravokeyl, 2017-04-18:
I was trying to use [UART-to-WebSocket](https://mongoose-os.com/uart-to-websocket.html) firmware but after the flashing the firmware, config `set` or `get` not working. I tried **step 3** which is like this

```
sudo ./mos config-set clubby.uart.uart_no=-1 uart.uart_no=0 debug.stderr_uart=-1 \
wifi.ap.enable=false wifi.sta.enable=true wifi.sta.ssid=NAME wifi.sta.pass=PASS
```
I'm getting the following error.

`Error: write /dev/ttyUSB0: file already closed`

The verbose log when I tried `sudo ./mos config-get wifi.sta --verbose --logtostderr`
```
Using port /dev/ttyUSB0
I0418 13:10:35.295181    1104 util.go:17] Using port /dev/ttyUSB0
I0418 13:10:35.295232    1104 serial.go:45] Opening /dev/ttyUSB0...
I0418 13:10:35.296084    1104 serial.go:55] /dev/ttyUSB0 opened: &{0xd7838fd0}, err: <nil>
I0418 13:10:40.301888    1104 serial.go:155] closing serial /dev/ttyUSB0
W0418 13:10:40.501970    1104 dev_conn.go:78] Error: context deadline exceeded
I0418 13:10:40.501885    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
W0418 13:10:40.502146    1104 dev_conn.go:78] Error: write /dev/ttyUSB0: file already closed
I0418 13:10:40.502165    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
I0418 13:10:40.502232    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
I0418 13:10:40.502290    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
I0418 13:10:40.502358    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
I0418 13:10:40.502304    1104 main.go:207] Error: write /dev/ttyUSB0: file already closed
write /dev/ttyUSB0: file already closed
/go/src/cesanta.com/common/go/mgrpc/codec/serial.go:123: 
/go/src/cesanta.com/common/go/mgrpc/codec/stream.go:287: 
/go/src/cesanta.com/common/go/mgrpc/mgrpc.go:281: 
/go/src/cesanta.com/fw/defs/config/config_service.go:68: 
/go/src/cesanta.com/mos/dev/dev_conn.go:81: 
/go/src/cesanta.com/mos/config.go:53: 
/go/src/cesanta.com/mos/main.go:120: 
I0418 13:10:40.502423    1104 mgrpc.go:237] error returned from codec Recv: read /dev/ttyUSB0: file already closed, keep trying
Error: write /dev/ttyUSB0: file already closed
```
What's the issue here? How can I get rid of `Error: context deadline exceeded`? I tried using `--tiemout` but it's not recognized.

anjeludo, 2017-05-03:
Same problem here.

closed 2017-05-30 by bravokeyl

caiobarbosa, 2017-06-08:
I have the same issue.
Anyone find out any solution?

heggenu, 2017-06-11:
Also same problem since two days back.

cpq, 2017-06-11:
Configuration commands `mos config-set` and `mos config-get`  are RPC commands.

And my default, `mos` uses UART channel for RPC, see https://mongoose-os.com/docs/core_components/rpc.html

Therefore, if you configure TCP-UART bridge, you take over the UART cause UART starts to be bridged, and RPC over serial naturally stop to work.

The solution is: after you configure TCPUART, use another RPC channel to talk to the device, e.g. websocket:

```
mos --port ws://IP_ADDRESS/rpc config-get
```