cesanta/mongoose-os #237: Support For MSP430 Boards

denzelwamburu, 2017-04-20:
Great work you guys are doing. 
Thanks for the OS 🎉 🎉 🎉 

Is there any way to port the OS to MSP430?
Or whats the procedure?

cpq, 2017-04-22:
MSP430 is too thin for Mongoose OS. only components like https://github.com/cesanta/mjs could be used.

MSP432 on the other hand could be a good target, but there are no immediate plans to do it.

Volunteers are welcome though.

denzelwamburu, 2017-04-22:
Thanks @cpq .
What about nrf52. They seem powerful enough.

denzelwamburu, 2017-04-22:
Would me amazing if you provide a doc on the process of porting the OS to other MCUs and supported MCU spec. Will provide a clear guidance to users who wanna port it, don't you think?

cpq, 2017-04-24:
nRF52 is certainly in the plans. We already have networking layer ported, https://www.hackster.io/dfrank/mongoose-embedded-networking-library-on-nrf51-and-nrf52-547b15

There are no instructions yet, but the code is open - https://github.com/cesanta/mongoose-os/tree/master/fw/platforms

Just follow an existing examples.

closed 2017-04-24 by cpq