cesanta/mongoose-os #238: Failed to load firmware

scargill, 2017-04-21:
I'm on Windows 10 and have a generic ESP8266 board currently flashing a light thanks to Arduino IDE - I also program the boards with ECLIPSE - so the board is a known worker on COM4.

I downloaded MOS to my desktop, created a MOS folder - ran from there... ../most --port COM4 flash mos-esp8266.   Actually I tried several approaches - this got the most flashing lights. I reset my board while holding GPIO0 low to put it into programming mode as usual (I'm using a separate FTDI).

..\mos --port COM4 flash mos-esp8266
Fetching https://mongoose-os.com/downloads/mos-esp8266.zip...
Loaded mjs_base/esp8266 version 1.0 (20170421-141840/???)
Opening COM4...
Connecting to ESP8266 ROM, attempt 1 of 10...
Connecting to ESP8266 ROM, attempt 2 of 10 (inverted)...
Connecting to ESP8266 ROM, attempt 3 of 10...
Connecting to ESP8266 ROM, attempt 4 of 10 (inverted)...
Connecting to ESP8266 ROM, attempt 5 of 10...
Connecting to ESP8266 ROM, attempt 6 of 10 (inverted)...
Connecting to ESP8266 ROM, attempt 7 of 10...
Connecting to ESP8266 ROM, attempt 8 of 10 (inverted)...
Connecting to ESP8266 ROM, attempt 9 of 10...
Connecting to ESP8266 ROM, attempt 10 of 10 (inverted)...
Error: failed to create ROM client: failed to connect to ROM: failed to connect to ESP8266 ROM

It does not matter how I do this - I've tried simply MOS which brings up a web interace - none of these will program the board. I can see the lights flashing on the FTDI - I know the board is in programming mode otherwise it would be running the blink example - it's an ESP12-E so no shortage of flash.

About to give up as I simply cannot get past this point.

Any ideas?

cpq, 2017-04-21:
Run with verbose debug, `mos --port COM4 flash esp8266 --v4 --logtostderr`
and show the logs please.
Also, are DTR & RTS are connected?

scargill, 2017-04-22:
Hi there

 

DTR and RTS are not used – many of us use boards with either separate reset and program or a combined (long push) button to reset the ESPs and maybe put them into programming mode – then it is merely necessary to hook up power, ground, RX and TX to do programming.

 

In order to use your software it is necessary to do something with RTS or DTR?  I believe they are accessible on the FTDI boards..

 

 

I tried

 

mos --port COM4 flash esp8266 --v4 –logtostderr

 

It would not run with –-v4 in it – so ran without the –-v4  don’t know where the log is ??

 

Also tried original with RTS hooked to ground – then to +V – made no difference.

 

Pete.

cpq, 2017-04-22:
Ah, apologies, it should be `--v 4 --logtostderr`

The error log should be dumped on stderr

scargill, 2017-04-22:
This is interesting - I used your line above with the fixed --v 4  - and it seems to have worked... there's a blue flashing light on the board  and the messages looked sensible.

See below - it looks like it has put the boot into the chip.

But when I go back to the graphical interface, select COM4 and hit Next (I have updated it as there was an update request in the Chrome Browser once I ran MOS) - same result - sits there busy... no flashing lights on the FTDI either...

I0422 11:16:32.079615   20296 report.go:13]     4096 @ 0x1000
I0422 11:16:32.079615   20296 flasher_client.go:108] FlashDigest(3) [4096 4096 0]
I0422 11:16:32.079615   20296 slip.go:91] => (1) 03
I0422 11:16:32.079615   20296 slip.go:91] => (12) 001000000010000000000000
I0422 11:16:32.093671   20296 slip.go:47] <= (16) 6ae59e64850377ee5470c854761551ea
I0422 11:16:32.093671   20296 flasher_client.go:135] <= "6ae59e64850377ee5470c854761551ea"
I0422 11:16:32.093671   20296 slip.go:47] <= (1) 00
I0422 11:16:32.093671   20296 flasher_client.go:135] <= "00"
  628224 @ 0x8000
I0422 11:16:32.093671   20296 report.go:13]   628224 @ 0x8000
I0422 11:16:32.093671   20296 flasher_client.go:108] FlashDigest(3) [32768 628224 0]
I0422 11:16:32.094674   20296 slip.go:91] => (1) 03
I0422 11:16:32.094674   20296 slip.go:91] => (12) 008000000096090000000000
I0422 11:16:32.381398   20296 slip.go:47] <= (16) d7dc51ec8f4bdd6614fc0fc2daa1d9bd
I0422 11:16:32.381398   20296 flasher_client.go:135] <= "d7dc51ec8f4bdd6614fc0fc2daa1d9bd"
I0422 11:16:32.383396   20296 slip.go:47] <= (1) 00
I0422 11:16:32.383396   20296 flasher_client.go:135] <= "00"
  131072 @ 0xdb000
I0422 11:16:32.385402   20296 report.go:13]   131072 @ 0xdb000
I0422 11:16:32.386399   20296 flasher_client.go:108] FlashDigest(3) [897024 131072 0]
I0422 11:16:32.386399   20296 slip.go:91] => (1) 03
I0422 11:16:32.386399   20296 slip.go:91] => (12) 00b00d000000020000000000
I0422 11:16:32.445546   20296 slip.go:47] <= (16) 80995d3985092865d9938ba6b4681d83
I0422 11:16:32.445546   20296 flasher_client.go:135] <= "80995d3985092865d9938ba6b4681d83"
I0422 11:16:32.447527   20296 slip.go:47] <= (1) 00
I0422 11:16:32.447527   20296 flasher_client.go:135] <= "00"
    4096 @ 0xfb000
I0422 11:16:32.447527   20296 report.go:13]     4096 @ 0xfb000
I0422 11:16:32.448528   20296 flasher_client.go:108] FlashDigest(3) [1028096 4096 0]
I0422 11:16:32.448528   20296 slip.go:91] => (1) 03
I0422 11:16:32.448528   20296 slip.go:91] => (12) 00b00f000010000000000000
I0422 11:16:32.461644   20296 slip.go:47] <= (16) 6ae59e64850377ee5470c854761551ea
I0422 11:16:32.461644   20296 flasher_client.go:135] <= "6ae59e64850377ee5470c854761551ea"
I0422 11:16:32.461644   20296 slip.go:47] <= (1) 00
I0422 11:16:32.461644   20296 flasher_client.go:135] <= "00"
     128 @ 0x3fc000
I0422 11:16:32.461644   20296 report.go:13]      128 @ 0x3fc000
I0422 11:16:32.461644   20296 flasher_client.go:108] FlashDigest(3) [4177920 128 0]
I0422 11:16:32.461644   20296 slip.go:91] => (1) 03
I0422 11:16:32.462650   20296 slip.go:91] => (12) 00c03f008000000000000000
I0422 11:16:32.476805   20296 slip.go:47] <= (16) b5e9562bfdf62840f5ca0eacc58809cd
I0422 11:16:32.477804   20296 flasher_client.go:135] <= "b5e9562bfdf62840f5ca0eacc58809cd"
I0422 11:16:32.478806   20296 slip.go:47] <= (1) 00
I0422 11:16:32.478806   20296 flasher_client.go:135] <= "00"
Booting firmware...
I0422 11:16:32.478806   20296 report.go:13] Booting firmware...
I0422 11:16:32.478806   20296 flasher_client.go:108] FlashBootFW(6) []
I0422 11:16:32.478806   20296 slip.go:91] => (1) 06
I0422 11:16:32.520619   20296 slip.go:47] <= (1) 00
I0422 11:16:32.520619   20296 flasher_client.go:135] <= "00"
All done!

scargill, 2017-04-22:
Incidentally I've tried MOS (which is sitting on my Windows 10 desktop - with a mos directory under there) both straight off the desktop and as administrator - makes no difference.

scargill, 2017-04-22:
Ok, by cording COM4 on the command line before running the UI - it works - I have a simple example working - but I can say for sure - on Windows 10 Pro - without that - the GUI seems unable to successfully select COM4

cpq, 2017-04-22:
Thank you.

If `--port` is not specified, then the UI enumerates all COM ports and makes a guess by choosing the very first port on the list. Do you have any other COM ports active?

Does the UI shows a correct port dropdown on the first wizard page?

scargill, 2017-04-22:
I do have other COM ports – several.  The UI shows a dropdown – always selecting COM10 (my camera) – so I drop down to COM4 and then continue – never works… but by running

 

mos –port COM4 ui

 

I seem to have this working…

 

I’ve just tested the simple example of a button triggering MQTT – using my own MQTT broker – that all seems to work… (had to put the port number in the address  - not sure if that is documented anywhere)..

cpq, 2017-04-22:
Selects COM10  by default? Hmm, I believe that's because of string-based search, COM10 in this case comes before COM4.

On the first UI page, you might select a correct port from the dropdown and click Next, that should work.

scargill, 2017-04-22:
Nope – it did not – I have everything working but only by selecting COM4 manually..

 

i.e. 

 

mos –port COM4 ui

cpq, 2017-04-22:
Got it, thank you Peter, we'll try to figure out why port enumeration fails.

cristiancs, 2017-05-01:
Hi, just want to add that in Mac it doesn't pick the port you select on the drop down, always use the first one.

cpq, 2017-05-01:
thanks @cristiancs , looking at it

cpq, 2017-06-01:
Fixed, please `mos update`.

closed 2017-06-01 by cpq