cesanta/mongoose-os #242: Crashes around DNS-SD caused by WiFi reconnects

kzyapkov, 2017-04-24:
With the latest mongoose-os (4388465b) we are seeing crashes on ESP8266 devices when they are forced to reconnect to a WiFi access point. These crashes occur sporadically, we've found that a reliable way to cause them is to connect them to an AP which gets constantly restarted (easy to accomplish on an OpenWRT/LEDE router by executing `wifi` periodically). An example traceback:

```

# excerpt from device log
ip:192.168.63.140,mask:255.255.255.0,gw:192.168.63.1

Exception 9 @ 0x4027aa5f, vaddr 0x612f6e45
 A0: 0x4027aa3f  A1: 0x3ffff6d0  A2: 0x612f6e3d  A3: 0x405e27ad
 A4: 0x00000007  A5: 0x00000078  A6: 0x7ff00000  A7: 0x7ffe27ad
 A8: 0x000003fe  A9: 0x3e000000 A10: 0x00000405 A11: 0xcfaacd9f
A12: 0x3ffefd6c A13: 0x000001f1 A14: 0x3fff1c54 A15: 0x000001e3

(exc SP: 0x3ffff530)

# backtrace from gdb

0x4027aa5f in mg_send (nc=nc@entry=0x3ffefd6c, buf=0x3fff1c54, len=497)
    at /mongoose-os/mongoose/mongoose.c:2453
2453	    nc->iface->vtable->tcp_send(nc, buf, len);
add symbol table from file "/mongoose-os/common/platforms/esp8266/rom/rom.elf" at
	.text_addr = 0x40000000
Reading symbols from /mongoose-os/common/platforms/esp8266/rom/rom.elf...(no debugging symbols found)...done.
(gdb) bt full
#0  0x4027aa5f in mg_send (nc=nc@entry=0x3ffefd6c, buf=0x3fff1c54, len=497)
    at /mongoose-os/mongoose/mongoose.c:2453
No locals.
#1  0x402765e0 in mg_dns_send_reply (nc=nc@entry=0x3ffefd6c, 
    r=r@entry=0x3fffff24) at /mongoose-os/mongoose/mongoose.c:10753
        sent = <optimized out>
#2  0x402723ed in dns_sd_advertise (c=c@entry=0x3ffefd6c)
    at /mongoose-os/fw/src/mgos_dns_sd.c:275
        mbuf1 = {buf = 0x3fff1c54 "\001\343", len = 497, size = 497}
        mbuf2 = {buf = 0x3ffefd64 "\023shellyswitch-1632BC\005local", 
          len = 30, size = 128}
        msg = {pkt = {p = 0x0, len = 0}, flags = 33920, transaction_id = 0, 
          num_questions = 0, num_answers = 6, questions = {{name = {p = 0x0, 
                len = 0}, rtype = 0, rclass = 0, ttl = 0, 
              kind = MG_DNS_INVALID_RECORD, rdata = {p = 0x0, 
                len = 0}} <repeats 32 times>}, answers = {{name = {p = 0x0, 
                len = 0}, rtype = 0, rclass = 0, ttl = 0, 
              kind = MG_DNS_INVALID_RECORD, rdata = {p = 0x0, len = 0}}, {
              name = {p = 0x0, len = 0}, rtype = 0, rclass = 0, ttl = 0, 
              kind = MG_DNS_INVALID_RECORD, rdata = {p = 0x0, len = 0}}, {
              name = {p = 0x40211a46 "", len = 0}, rtype = 33, rclass = 32769, 
              ttl = 120, kind = MG_DNS_ANSWER, rdata = {p = 0x0, len = 0}}, {
              name = {p = 0x0, len = 0}, rtype = 0, rclass = 0, ttl = 0, 
---Type <return> to continue, or q <return> to quit---
              kind = MG_DNS_INVALID_RECORD, rdata = {p = 0x0, 
                len = 0}} <repeats 29 times>}}
        reply = {msg = 0x3ffff710, io = 0x3fffff3c, start = 0}
        __func__ = "dns_sd_advertise"
#3  0x402724a0 in dns_sd_wifi_ev_handler (event=MGOS_WIFI_IP_ACQUIRED, 
    data=0x0) at /mongoose-os/fw/src/mgos_dns_sd.c:293
        c = 0x3ffefd6c
        __func__ = "dns_sd_wifi_ev_handler"
#4  0x4028036e in mgos_wifi_on_change_cb (event=MGOS_WIFI_IP_ACQUIRED)
    at /mongoose-os/fw/src/mgos_wifi.c:51
        __func__ = "mgos_wifi_on_change_cb"
        e = <optimized out>
        te = 0x0
#5  0x40273908 in invoke_wifi_on_change_cb (arg=<optimized out>)
    at /mongoose-os/fw/platforms/esp8266/src/esp_wifi.c:28
No locals.
#6  0x4027e51f in mgos_lwip_task (e=<optimized out>)
    at /mongoose-os/fw/platforms/esp8266/src/esp_main.c:224
        cb = <optimized out>
#7  0x40000f49 in ets_post ()
```

cpq, 2017-04-24:
Do you have dns_sd enabled?

kzyapkov, 2017-04-24:
Yes, and it normally works fine.

kzyapkov, 2017-05-02:
@cpq @rojer are you able to reproduce this on your end?

closed 2017-05-15 by rojer