cesanta/mongoose-os #243: Disabling network service discovery breaks esp8266 build

neoautomata, 2017-04-25:
Setting the following in `mos.yml` breaks the build:
```
build_vars:
  MGOS_ENABLE_DNS_SD: 0
```
```
$ mos build
...
While parsing /mongoose-os/fw/platforms/esp8266/src/esp_sys_config.yaml: 'dns_sd.host_name: Cannot override, no such entry'While parsing /mongoose-os/fw/platforms/esp8266/src/esp_sys_config.yaml: 'dns_sd.host_name: Cannot override, no such entry'
...
```

I am not setting or using `dns_sd.host_name` in my code.

neoautomata, 2017-04-25:
Forgot to note that if I set `MGOS_ENABLE_DNS_SD` to `1` , then the build works.

rojer, 2017-04-25:
fixed by https://github.com/cesanta/mongoose-os/commit/721722e6f0f221a96e3026c89e76efcd811b2dbb

closed 2017-04-25 by rojer