cesanta/mongoose-os #245: JSON_OUT_MBUF breaks C++ build

neoautomata, 2017-04-26:
I'm trying to use C++, but the `JSON_OUT_MBUF` macro expansion doesn't compile. The code is:
```cpp
struct mbuf fb;
struct json_out out = JSON_OUT_MBUF(&fb);
mbuf_init(&fb, 32);
```

Which causes the following error with `mos build`
```
In file included from src/main.cpp:20:0:
src/main.cpp: In function 'void relay_handler(mg_rpc_request_info*, void*, mg_rpc_frame_info*, mg_str)':
/mongoose-os/common/json_utils.h:29:3: error: invalid conversion from 'void*' to 'char*' [-fpermissive]
   }
   ^
src/main.cpp:61:24: note: in expansion of macro 'JSON_OUT_MBUF'
  struct json_out out = JSON_OUT_MBUF(&fb);
                        ^
```

neoautomata, 2017-04-26:
I think just adding an explicit cast to the macro will fix it. I tried manually expanding it and it compiles as either a `.c` or `.cpp`.

Current version:
```cpp
#define JSON_OUT_MBUF(mbuf_addr)   \
  {                                \
    mg_json_printer_mbuf, {        \
      { (void *) mbuf_addr, 0, 0 } \
    }                              \
  }
```

Explicit cast:
```cpp
#define JSON_OUT_MBUF(mbuf_addr)             \
  {                                          \
    mg_json_printer_mbuf, {                  \
      { (char *)((void *) mbuf_addr), 0, 0 } \
    }                                        \
  }
```

neoautomata, 2017-05-09:
Instead of the double cast, it should just be a cast to `char *`. I've been working around it with the following after my includes:
```cpp
#undef JSON_OUT_MBUF
#define JSON_OUT_MBUF(mbuf_addr)             \
  {                                          \
    mg_json_printer_mbuf, {                  \
      { (char *)mbuf_addr, 0, 0 } \
    }                                        \
  }
```

neoautomata, 2017-05-09:
I just noticed [issue 241](https://github.com/cesanta/mongoose-os/issues/241) which appears to be the same issue and was filed a few days before.

Feel free to dedupe if needed.

neoautomata, 2017-05-17:
I've sent [Pull Request 263](https://github.com/cesanta/mongoose-os/pull/263) to fix this.

cpq, 2017-06-01:
Duplicate of #241 

Ruslan, please take care

ruslanvaliullin, 2017-06-02:
done.

closed 2017-06-02 by ruslanvaliullin