cesanta/mongoose-os #247: Is this thing working on ESP8266-01 or my efforts are futile?

huseyint, 2017-04-27:
I am running on Windows and using and FTDI adapter to connect my ESP8266-01 module to my PC. Flashing works flawlessly (I am connecting GPIO0 to GND while flashing) but I cannot proceed, neither I can set the wifi configuration on web UI, nor I can do any sort of ls stuff.

Here is my flash log:

```
C:\Users\huseyin>mos --port COM6 flash mos-esp8266
Fetching https://mongoose-os.com/downloads/mos-esp8266.zip...
Loaded mjs_base/esp8266 version 1.0 (20170426-154924/???)
Opening COM6...
Connecting to ESP8266 ROM, attempt 1 of 10...
  Connected
Running flasher @ 460800...
  Flasher is running
Flash size: 1048576, params: 0x0220
Deduping...
    2544 @ 0x0 -> 0
  632320 @ 0x8000 -> 0
  131072 @ 0xdb000 -> 0
     128 @ 0xfc000 -> 0
Writing...
    4096 @ 0x1000
    4096 @ 0xfb000
Wrote 8192 bytes in 0.22 seconds (286.15 KBit/sec)
Verifying...
    2544 @ 0x0
    4096 @ 0x1000
  632320 @ 0x8000
  131072 @ 0xdb000
    4096 @ 0xfb000
     128 @ 0xfc000
Booting firmware...
All done!
```

This is what I get if I try to set wifi config (first one is GPIO0 connected to GND, second not):

```
C:\Users\huseyin>mos wifi SSID PASSWORD
Using port COM6
Getting configuration...
Error: The handle is invalid.

C:\Users\huseyin>mos wifi SSID PASSWORD
Using port COM6
Getting configuration...
Error: context deadline exceeded
```

And this is what I get when I run the `console` command:

```
C:\Users\huseyin>mos --port COM6 console
r   l  r   c  n            p |     x         p    nn   ; n       b $ r  r  p n          l         b  n  n $      b  >~ n       l`    # n   r    nr   ;       ;  p n       r          p  p  <       b  >~ n      r     # n   r    nr   ;     l r   r   n
rBoot v1.2.1 - richardaburton@gmail.com
Flash Size:   8 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz
rBoot Option: Big flash

Booting rom 0.
     8  n  > r   n b   l  l` > r x   mode : softAP(a2:20:a6:04:23:5b)
add
esp_mgos_init2       mjs_base 1.0 (20170426-154924/???)
esp_mgos_init2       Mongoose OS Firmware 2017042615 (20170426-154924/???)
esp_mgos_init2       SDK 2.0.0(656edbf), RAM: 51968 total, 49548 free
esp_print_reset_info Reset cause: 6 (sys reset)
fs_moubcn 0
del if1
usl
mode : null
mode : softAP(a2:20:a6:04:23:5b)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
usl
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
nt             Mounting FS: 131072 @ 0xdb000
mgos_sys_config_init MAC: A220A604235B
mgos_sys_config_init WDT: 30 seconds
mgos_wifi_set_mode   WiFi mode: AP
mgos_wifi_setup_ap   WiFi AP: SSID Mongoose_04235B, channel 6
mgos_wifi_setup_ap   WiFi AP IP: 192.168.4.1/255.255.255.0 gw 192.168.4.1, DHCP range 192.168.4.2 - 192.168.4.100
mgos_i2c_create      I2C GPIO init ok (SDA: 12, SCL: 14)
mgos_sys_config_init HTTP server started on [80]
mg_rpc_channel_mqtt  0x3fff08dc esp8266_04235B/rpc/#
mg_rpc_channel_uart
```

Any ideas?

cristiancs, 2017-04-29:
try switch to protopying>configuration> find dns_sd and change  "host_name": "mOS_esp8266_????" with something random like "mOS_esp8266_7CiCH"

cpq, 2017-04-29:
no, there is no chance to fit mOS on esp01.

mOS uses about 700+k of flash.

closed 2017-04-29 by cpq

denzelwamburu, 2017-05-02:
Hi, am using the WEB ui and its returning Error 4.
Tried on both esp82 and sparkfun thing.

cpq, 2017-05-02:
run as 
```
mos flash esp8266 --v 4 --logtostderr
```

paste the log

denzelwamburu, 2017-05-02:
I0502 13:02:12.888599   27950 main.go:211] Error: flash size is not specified and could not be detected: invalid chip id: 0x001f8401
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher/common.go:101: invalid chip id: 0x001f8401
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher/common.go:67: flash size is not specified and could not be detected
/Users/lsm/src/cesanta.com/mos/flash/esp/flasher/flash.go:43:
/Users/lsm/src/cesanta.com/mos/flash.go:125:
/Users/lsm/src/cesanta.com/mos/main.go:124:

denzelwamburu, 2017-05-02:
Getting another error.
Am i supposed to specify the size?

cpq, 2017-05-02:
Could you move this dicussion to forum.mongoose-os.com please?