cesanta/mongoose-os #249: WDT Resets and ESP32 Hangs on "Pro cpu start user code"

jkachmar, 2017-05-01:
Not entirely sure what the best way to report this is, as I don't have a consistent repro. 

I'm effectively running a polling loop on an ESP32 that regularly reports in to AWS IoT using the provided C frameworks. On the ESP8266 this works as-expected, however on the ESP32 it appears as if some system events trigger the WDT and Mongoose OS resets into some mode that prevents my code from executing.

I'm including some logs in the gist below, and can try to reproduce with a different log-level if necessary. Unfortunately it comes and goes, some devices have been fine for days, others lock up shortly after reset. I've tested this on the standard ESP32 dev board as well as Sparkfun's ESP32 Thing so far.

https://gist.github.com/jkachmar/ecce6ea1dce02e614dfd99af64f6ba76

cpq, 2017-05-03:
We need to see your code.

jkachmar, 2017-05-03:
I've extracted out a small repro here, based on the c_aws_shadow example:

https://gist.github.com/jkachmar/ca0266073b9a21bfb22af4921743b552

The polling of GPIO 0 (vs. triggering an interrupt) is intentional to mirror other behavior. 

Curiously, with _just_ the GPIO polling, I was unable to reproduce the crash; it was only after I added in a call to `mgos_adc_read()` that it ended up crashing again.

rojer, 2017-05-09:
sorry, i tried your code and was unable to repro - the loop runs fine for me, with and without mgos_adc_read, for at least several minutes. no WDT resets and no boot hangs.
can you still reproduce it? do you have another module to try it on?

jkachmar, 2017-05-11:
Thanks for looking into it, I've since bumped my Mongoose OS build environment and the problem _seems_ to have resolved itself.

For reference, I was running into this on [4439978](https://github.com/cesanta/mongoose-os/tree/4439978b7aed8f54fe9fb7282993cfccb0033075), and I'm currently on [5a3a3af](https://github.com/cesanta/mongoose-os/tree/5a3a3af14287837a46675b3357b98f5e3a458bce).

Feel free to close this issue if I'm the only one reporting this sort of problem; if I run into it again I can just reopen, hopefully with additional details :smile:

closed 2017-05-11 by cpq