cesanta/mongoose-os #253: How to work in multi-core mode on ESP32?

aifer2007, 2017-05-04:
ESP32 is multi-core MCU.

In multi-core mode, the booting info is :

I (1403) cpu_start: Pro cpu up.
I (1415) cpu_start: Starting app cpu, entry point is 0x40080bb4
I (0) cpu_start: App cpu up.
I (1447) cpu_start: Pro cpu start user code
I (1639) phy: phy_version: 258, Nov 29 2016, 15:51:07, 0, 0
I (2334) cpu_start: Starting scheduler on PRO CPU.
starting app_main()
I (905) cpu_start: Starting scheduler on APP CPU.

Tow cores start: Pro cpu and App cpu.

Mongoose-os booting info :

I (1636) cpu_start: Pro cpu up.
I (1647) cpu_start: Single core mode
I (1831) cpu_start: Pro cpu start user code
I (1886) cpu_start: Starting scheduler on PRO CPU.
Only one core is start: Pro cpu up.

Can mongoose-os work in multi-core mode? And how?

cpq, 2017-05-09:
See https://forum.mongoose-os.com/index.php?p=/discussion/comment/2356/#Comment_2356

closed 2017-05-09 by cpq

brandond, 2018-01-02:
Can we reopen this until dual-core is supported? Half of the available processing power of this chipset is going to waste. More, if you count the unsupported ULP core.

tripflex, 2019-02-15:
@brandond @aifer2007 https://github.com/cesanta/mongoose-os/issues/480