cesanta/mongoose-os #258: HTTP.query not working for HTTPS requests

andreicek, 2017-05-16:
Hi,

the following code will result in no HTTPS calls made. If I use HTTP there are no problems what so ever.

```javascript
load('api_http.js');
load('api_gpio.js');

let pin = 0;
GPIO.set_button_handler(pin, GPIO.PULL_UP, GPIO.INT_EDGE_NEG, 200, function(x) {
  print('starting network request');
  HTTP.query({
    url: 'https://js101.byinfinum.co/hello.json',
    success: function(res) {
      print('network request done');
      print(res);
    }
  });
}, true);
```

I've tried debugging `api_http.js` and it seems that the function call `this._cs(obj.u.addr, f, obj, '', '');` does nothing, or to be precise that `_cs: ffi('void *mgos_connect_http_ssl(char *, void (*)(void *, int, void *, userdata), userdata, char *, char *)')` is not working at all.

Thanks!

cpq, 2017-05-16:
Thank you, we're working on it already!

ruslanvaliullin, 2017-05-17:
Fixed https://github.com/cesanta/mongoose-os/commit/f1f7309020e3efabb7cd9d513c2d01f927f2f6e2. Please check. Regards.

closed 2017-05-17 by ruslanvaliullin

andreicek, 2017-05-18:
Thanks! Can you please tell me when this will be available in the release?

ruslanvaliullin, 2017-05-18:
You can already use it. 
https://github.com/cesanta/mongoose-os/blob/master/fw/examples/mjs_base/examples/https_client.js

andreicek, 2017-05-18:
When I update thru the wizard I don't get the updated files. This makes me think that the firmware package didn't get updated: `https://mongoose-os.com/downloads/mos-esp32.zip`.

[image]

Thanks for the help.

cpq, 2017-05-18:
@andreicek loading/importing examples is not yet well done in mOS - work in progress.

For now, flash default firmware, open init.js, copy/paste example code, save&reboot.
Example code:

```
// Load Mongoose OS API
load('api_gpio.js');
load('api_http.js');

// On nodemcu, press flash button to send HTTPS request
GPIO.set_button_handler(0, GPIO.PULL_UP, GPIO.INT_EDGE_NEG, 200, function() {
  print('sending request...');
  HTTP.query({
    url: 'https://httpbin.org/get',
    success: function(body, full_http_msg) { print(body); },
    error: function(err) { print(err); }  // Optional
  });
}, null);
```

Please let us know how it goes.

andreicek, 2017-05-18:
Still nothing. I'm confused as to how the `api_http.js` file should be updated. I tried running `~/.mos/bin/mos update` but I keep getting Up to date message.

Currently, on my device, the `api_http.js` is different from the one in the mentioned commit.

cpq, 2017-05-18:
Ah, yes, you're talking about api_http.js.
Firmware updated.

andreicek, 2017-05-18:
Working! Thanks for all the help and fast reaction! 🎉

mounikamadurireddy, 2017-09-23:
hi,
iam new to mongoose os .
Can i get any code for esp in http get and post data

aleksmarcelo, 2017-10-09:
Still not working when you use the IP address

gnanaswaroop, 2017-10-31:
> https://github.com/cesanta/mongoose-os/blob/master/fw/examples/mjs_base/examples/https_client.js

This link is broken now. Couldn't find it in archives either. Would someone be kind enough to post the correct URL?

I downloaded the latest MOS and am testing on the latest version.

bravokeyl, 2017-10-31:
@gnanaswaroop , What exactly are you trying to do, I think you are posting on a wrong issue/thread. Anyway, the above link/code is migrated to the apps section which should be [Default App](https://github.com/mongoose-os-apps/default) AFAIK

gnanaswaroop, 2017-10-31:
@bravokeyl I am trying to establish a HTTPS connection and am looking for an example to do that. I see your mention of the Default App but cant seem to find the source/example inside that.

bravokeyl, 2017-10-31:
If you are looking for HTTP based example, here is [one](https://github.com/mongoose-os-apps/http-fetch)

gnanaswaroop, 2017-10-31:
@bravokeyl I am actually looking for an HTTPS based example using MOS via Javascript to which the answers appear that something was fixed in MOS that resolved the HTTPs as a client connection to server.

The provided example is merged into some default app but can't seem to find the example there either.

gnanaswaroop, 2017-11-06:
@andreicek Did u have any luck with the HTTPS client calls from the IoT device?

gnanaswaroop, 2017-11-06:
@andreicek Can you please post which hardware you were trying on? I am trying to run this on an NodeMCU ESP8266

gnanaswaroop, 2017-11-08:
Found the source although the link was broken. The example is the same that @cpq mentioned inline above. 

https://github.com/cesanta/mongoose-os/blob/f1f7309020e3efabb7cd9d513c2d01f927f2f6e2/fw/examples/mjs_base/examples/https_client.js

phatpaul, 2020-03-13:
Thanks for digging this up @gnanaswaroop 
I can't believe it was so hard to find an example of a simple HTTP request using mjs.  BTW, anyone know where's the API documentation on that function?

nliviu, 2020-03-13:
https://github.com/mongoose-os-libs/mjs/blob/master/fs/api_http.js#L52-L73