cesanta/mongoose-os #260: CC3200 crashes when switching WiFi mode

kzyapkov, 2017-05-17:
On CC3200, when flipping between AP and STA modes, the device crashes, apparently due to memory corruption. This appeared recently, but we cannot pinpoint the change causing the issue.

The crash does not happen every time WiFi mode is changed, but is frequent enough.

Some sample code to demonstrate the issue:

```
void flip_wifi(void *arg) {
    (void) arg;
    struct sys_config *cfg = get_cfg();
    if (cfg->wifi.ap.enable) {
        cfg->wifi.ap.enable = false;
        cfg->wifi.sta.enable = true;
        mgos_wifi_setup_sta(&cfg->wifi.sta);
    } else {
        cfg->wifi.ap.enable = true;
        cfg->wifi.sta.enable = false;
        mgos_wifi_setup_ap(&cfg->wifi.ap);
    }
    save_cfg(cfg, NULL);
}

enum mgos_app_init_result mgos_app_init(void) {
    struct sys_config *cfg = get_cfg();
    flip_wifi(NULL);
    mgos_set_timer(5000, 1, flip_wifi, NULL);

    return MGOS_APP_INIT_SUCCESS;
}
```

For this to work, `conf_schema.yaml` should contain valid, working `wifi.sta` configuration. Within a few minutes, most probably when switching to STA mode, you will see:

```
--- Bus Fault ---
  SHCTL=0x00070002, FSTAT=0x00008200, HFSTAT=0x00000000, FADDR=200a95a4
  SF @ 0x2002c058:
    R0=0x200295ac R1=0x0000ffff R2=0x00000004 R3=0xfffffff5 R12=0x2002ccc8
    LR=0x200192e3 PC=0x200192ee xPSR=0x01000000
---
```

rojer, 2017-05-26:
thanks Kirill for a detailed report, i found and fixed the issue, fix will land soon

closed 2017-05-26 by rojer

rojer, 2017-05-26:
also, minor optimization along the way: https://github.com/cesanta/mongoose-os/commit/3946ee7b96280af7714055784e905172fd64000d

kzyapkov, 2017-05-26:
Thanks!

Quick question: is there a way to attach a debugger on this target, or look at core dumps like on the ESP8266?

rojer, 2017-05-26:
unfortunately not, we never ported coredumps to CC3200.