cesanta/mongoose-os #261: Windows Build fail

neoautomata, 2017-05-17:
C:\Users\User>mkdir mos-test

C:\Users\User>cd mos-test

C:\Users\User\mos-test>mos init --arch esp8266
Connecting to http://mongoose.cloud, user  ...
Downloading project skeleton...
Unpacking...
Setting arch "esp8266"...

C:\Users\User\mos-test>mos build
Connecting to http://mongoose.cloud, user test
Uploading sources (1496 bytes)
The flag --repo is not given, going to use mongoose-os repository
Sources: [src/*.c src/*.cpp]
Building...
Docker arguments: run --rm -i -v /tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774:/app -v /tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774/mongoos
e-os:/mongoose-os -v /tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774/mongoose-os:/tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774/mongoose-os -v
/tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774/src:/tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774/src -v /tmp/fwbuild-volumes/users/test/mos-t
est/build_requests/build_req_969578774:/tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_969578774 --user 0:0 docker.cesanta.com/esp8266-build:2.0.0-1.5.0-r4 /bin/bash -c nice make '-j
' '16' '-C' '/app' '-f' '/mongoose-os/fw/platforms/esp8266/Makefile.build' 'FW_DIR=build/fw' 'GEN_DIR=build/gen' 'FFI_SYMBOLS=' 'PLATFORM=esp8266' 'BUILD_DIR=build/objs' 'APP_SOURCES=src/*.c src/*.cpp
' 'APP_CONF_SCHEMA=src/conf_schema.yaml' 'APP_VERSION=1.0' 'APP_FS_FILES=fs' 'MGOS_PATH=/mongoose-os' 'FS_STAGING_DIR=build/fs' 'APP=mos-test'
make: Entering directory '/app'
  MKDIR /app/build/fw
  MKDIR /app/build/objs/fw_temp
  MKDIR /app/build/objs
  MKDIR /app/build/gen
CC esptool2.c
gcc -O2 -Wall -c esptool2.c -o /app/build/objs/esptool2.o
  GEN   /app/build/gen/sys_config.c
CC esptool2_elf.c
gcc -O2 -Wall -c esptool2_elf.c -o /app/build/objs/esptool2_elf.o
  GEN   /app/build/gen/sys_ro_vars.c
  GEN   /app/build/gen/sys_config_schema.json
  GEN   /app/build/gen/sys_ro_vars_schema.json
make[1]: Entering directory '/app'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections     -o .o
make: warning: undefined variable 'TARGET_ARCH'
make: warning: undefined variable 'LOADLIBES'
make: warning: undefined variable 'LDLIBS'
  GEN   /mongoose-os/fw/platforms/esp8266/ld/non_os.ld (0xd2ff0 @ 0x40208010) -> /app/build/gen/0x8000.ld
  CC    ./src/core/stats.c -> /app/build/objs/lwip/stats.o
  CC    ./src/netif/etharp.c -> /app/build/objs/lwip/etharp.o
  CC    ./src/core/mem.c -> /app/build/objs/lwip/mem.o
make[1]: Entering directory '/app'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  GEN   /app/build/objs/fw_temp/0x01000.bin
  GEN   /app/build/objs/fw_temp/0xfb000.bin
  CC    atca_command.c
/opt/Espressif4096+0 records in
8+0 records out
/4096 bytes (4.1 kB, 4.0 KiB) copied, 0.0015349 s, 2.7 MB/s
esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../4096+0 records in
8+0 records out
../../../xtensa-lx106-elf/bin/ld: no input files
4096 bytes (4.1 kB, 4.0 KiB) copied, 0.0016365 s, 2.5 MB/s
collect2: error: ld returned 1 exit status
make: *** [.o] Error 1
make: *** Waiting for unfinished jobs....
<builtin>: recipe for target '.o' failed
  CC    ./src/core/mdns.c -> /app/build/objs/lwip/mdns.o
  CC    ./src/core/sys_arch.c -> /app/build/objs/lwip/sys_arch.o
  CC    ./src/core/init.c -> /app/build/objs/lwip/init.o
  CC    ./src/core/memp.c -> /app/build/objs/lwip/memp.o
  CC    atca_iface.c
LD /app/build/objs/esptool2
gcc -o /app/build/objs/esptool2 /app/build/objs/esptool2.o /app/build/objs/esptool2_elf.o
  CC    ./src/core/tcp_in.c -> /app/build/objs/lwip/tcp_in.o
  CC    ./src/core/dhcp.c -> /app/build/objs/lwip/dhcp.o
  CC    ./src/core/dns.c -> /app/build/objs/lwip/dns.o
  CC    atca_cfgs.c
  CC    atca_device.c
  CC    basic/atca_helpers.c
  CC    ./src/core/raw.c -> /app/build/objs/lwip/raw.o
  CC    ./src/core/tcp.c -> /app/build/objs/lwip/tcp.o
  CC    basic/atca_basic.c
  CC    ./src/core/sntp.c -> /app/build/objs/lwip/sntp.o
  CC    ./src/core/def.c -> /app/build/objs/lwip/def.o
  CC    ./src/core/netif.c -> /app/build/objs/lwip/netif.o
  CC    ./src/core/sys.c -> /app/build/objs/lwip/sys.o
  CC    ./src/core/timers.c -> /app/build/objs/lwip/timers.o
  CC    ./src/core/pbuf.c -> /app/build/objs/lwip/pbuf.o
  CC    ./src/core/udp.c -> /app/build/objs/lwip/udp.o
  CC    ./src/core/tcp_out.c -> /app/build/objs/lwip/tcp_out.o
  CC    ./src/core/ipv4/ip_frag.c -> /app/build/objs/lwip/ip_frag.o
  CC    ./src/core/ipv4/inet.c -> /app/build/objs/lwip/inet.o
  CC    ./src/core/ipv4/ip_addr.c -> /app/build/objs/lwip/ip_addr.o
  CC    ./src/core/ipv4/igmp.c -> /app/build/objs/lwip/igmp.o
  CC    ./src/core/ipv4/ip.c -> /app/build/objs/lwip/ip.o
  CC    host/atca_host.c
  CC    ./src/core/ipv4/autoip.c -> /app/build/objs/lwip/autoip.o
  CC    ./src/core/ipv4/inet_chksum.c -> /app/build/objs/lwip/inet_chksum.o
  CC    ./src/core/ipv4/icmp.c -> /app/build/objs/lwip/icmp.o
  CC    ./src/api/netbuf.c -> /app/build/objs/lwip/netbuf.o
  CC    ./src/api/api_lib.c -> /app/build/objs/lwip/api_lib.o
  CC    ./src/api/netifapi.c -> /app/build/objs/lwip/netifapi.o
  CC    ./src/api/tcpip.c -> /app/build/objs/lwip/tcpip.o
  CC    ./src/api/netdb.c -> /app/build/objs/lwip/netdb.o
  CC    ./src/api/err.c -> /app/build/objs/lwip/err.o
  CC    ./src/api/api_msg.c -> /app/build/objs/lwip/api_msg.o
  CC    ./src/api/sockets.c -> /app/build/objs/lwip/sockets.o
  AR    /app/build/objs/atca/libatca.a
/opt/Espressif/crosstool-NG/builds/xtensa-lx106-elf/bin/xtensa-lx106-elf-ar -rc /app/build/objs/atca/libatca.a /app/build/objs/atca/atca_command.o /app/build/objs/atca/atca_iface.o /app/build/objs/atc
a/atca_cfgs.o /app/build/objs/atca/atca_device.o /app/build/objs/atca/atca_helpers.o /app/build/objs/atca/atca_basic.o /app/build/objs/atca/atca_host.o
make[1]: Leaving directory '/mongoose-os/third_party/cryptoauthlib/lib'
  CC    ./espressif/app/espconn.c -> /app/build/objs/lwip/espconn.o
  CC    ./espressif/app/espconn_buf.c -> /app/build/objs/lwip/espconn_buf.o
  CC    ./espressif/app/netio.c -> /app/build/objs/lwip/netio.o
  CC    ./espressif/app/espconn_tcp.c -> /app/build/objs/lwip/espconn_tcp.o
  CC    ./espressif/app/ping.c -> /app/build/objs/lwip/ping.o
  CC    ./espressif/app/dhcpserver.c -> /app/build/objs/lwip/dhcpserver.o
  CC    ./espressif/app/espconn_mdns.c -> /app/build/objs/lwip/espconn_mdns.o
  CC    ./espressif/app/espconn_udp.c -> /app/build/objs/lwip/espconn_udp.o
  AR /app/build/objs/lwip/liblwip_cs.a
make[1]: Leaving directory '/mongoose-os/common/platforms/esp8266/lwip'
make: Leaving directory '/app'
Error: exit status 2
Error: build failed

C:\Users\User\mos-test>

wuyuanyi135, 2017-05-24:
I have the same issue.
```
PS C:\Users\wuyua\test_mos> mos version
The Mongoose OS command line tool, v. 20170524-140954/master@41976534.
```

michaeldmoore, 2017-05-26:
Me too, with the next (current, as of writing) version of mos

C:\Users\mdmoo\foo>mos version
The Mongoose OS command line tool, v. 20170525-152605/master@f5603d7d.

Also, after seeing the warning message: 
"WARNING: Setting build variable "APP_CONF_SCHEMA" in "mos.yml" is deprecated, use "config_schema" property instead."

manually updated the mos.yml to:

[image]

neoautomata, 2017-05-30:
Confirmed a new build attempt on Linux fails. My existing projects all still build though. The build log also contains:

```
  GEN   /app/build/gen/sys_ro_vars_schema.json
make[1]: Entering directory '/app'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
```

I'm not sure how to toggle this behavior in a new project though.

neoautomata, 2017-05-30:
I spoke too soon, there are multiple occurrences. From my existing project:

```
make: Entering directory '/app'
  MKDIR /app/build/fw
  MKDIR /app/build/objs
  MKDIR /app/build/gen
  MKDIR /app/build/objs/fw_temp
  GEN   /app/build/gen/sys_config.c
CC esptool2.c
gcc -O2 -Wall -c esptool2.c -o /app/build/objs/esptool2.o
CC esptool2_elf.c
  GEN   /app/build/gen/sys_ro_vars.c
gcc -O2 -Wall -c esptool2_elf.c -o /app/build/objs/esptool2_elf.o
  GEN   /app/build/gen/sys_config_schema.json
  GEN   /app/build/gen/sys_ro_vars_schema.json
make[1]: Entering directory '/app'
make[1]: warning: -jN forced in submake: disabling jobserver mode.
  GEN   /mongoose-os/fw/platforms/esp8266/ld/non_os.ld (0xd2ff0 @ 0x40208010) -> /app/build/gen/0x8000.ld
  CC    ./src/netif/etharp.c -> /app/build/objs/lwip/etharp.o
  CC    ./src/core/stats.c -> /app/build/objs/lwip/stats.o
  CC    ./src/core/mem.c -> /app/build/objs/lwip/mem.o
make[1]: Entering directory '/app'
make[1]: warning: jobserver unavailable: using -j1.  Add '+' to parent make rule.
  CC    atca_command.c
...
```

neoautomata, 2017-05-30:
If I copy `mos.yml` and the `src` directory from the existing project, it still fails.

It seems that the `mos init` is no longer including an expected SSL cert under `fs/`. I created a new project in an empty directory with `mos init --arch esp8266` and attempt a build, which failed. I then copied `fs/DST_Root_CA_X3.pem` from my existing project into the empty `fs/` in the new project. The build then succeeds.

michaeldmoore, 2017-05-30:
I confirm this - I grabbed a copy of DST_Root_CA_x3.pem from github, placed it in the fs directory, and now the build runs to completion without errors.

neoautomata, 2017-05-30:
It doesn't have to be the `DST_Root_CA_x3.pem`, any file in `fs/` will have the build succeed.

```shell
$ mkdir test/

$ cd test/

$ mos init --arch esp8266
Connecting to http://mongoose.cloud, user  ...
Downloading project skeleton...
Unpacking...
Setting arch "esp8266"...

$ mos build
...build error...

$ touch fs/uhbadf

$ mos build
===
WARNING: Setting build variable "APP_CONF_SCHEMA" in "mos.yml" is deprecated, use "config_schema" property instead.
===
Connecting to http://mongoose.cloud, user test
Uploading sources (1632 bytes)
Success, built test/esp8266 version 1.0 (20170530-235452/???).
Firmware saved to build/fw.zip
```

dimonomid, 2017-05-31:
It's fixed now, thanks for reporting!

closed 2017-05-31 by dimonomid