cesanta/mongoose-os #266: WebUI unusable when stuck in restart cycle

scotthernandez, 2017-05-21:
When a mistakes is made in the init.js file that causes the device to restart (in a loop) then the web ui can no longer be used to update/replace the init.js file, to fix the problem.

This is because the web UI now tries to "reconnect" with a modal dialog on top of the editor. This makes the editor and "save" buttons unusable since every time the device restarts the dialog pops up again...

This is a poor user-experience compared to the old connection wizard method.

I'm happy to talk about possible solutions but I suspect that the modal nature of the connection box is going to be problematic if it automatically pops up without the user clicking something, no matter how it is coded. Maybe an "auto-reconnect" checkbox should be added (which disables this behavior), or some additional UI element to indicate the broken connection state and to trigger the dialog.

cpq, 2017-05-31:
Maybe the dialog should be triggered only once at startup ?

cpq, 2017-06-01:
`mos update` please.

Now, a dialog only pops up once at startup, only if device is not connected.

Note: together with this push, an initial support for apps/libs is also pushed.
Very fresh, so bugs / hiccups are expected. Please be forgiving !

Issue reports are appreciated as always.

Thank you.

cpq, 2017-06-01:
@scotthernandez you might be interested in that too:

https://mongoose-os.com/blog/mongoose-os-google-iot-esp8266/
https://mongoose-os.com/blog/mongoose-os-google-iot-ecc508a/

closed 2017-06-01 by cpq