cesanta/mongoose-os #267: coredump on ESP8266 when sending UDP on missing network

ceco, 2017-05-22:
Hi.

We get a coredump when trying to send UDP packet while connected in STA mode and there is no Internet connection.

Reproduce with the following example:
main.c:
```
#include "common/platform.h"
#include "fw/src/mgos_app.h"
#include "fw/src/mgos_mongoose.h"
#include "fw/src/mgos_sys_config.h"
#include "fw/src/mgos_wifi.h"
#include "fw/src/mgos_timers.h"
#include "frozen/frozen.h"

static const char* str_connect_uri = "udp://pool.ntp.org:123";
static struct mg_connection* sntp_nc = NULL;
static char str_sntp_request[48];

// Main network connection handler
void sntp_connection_handler(struct mg_connection *nc, int ev, void *ev_data, void *user_data) {
    (void) user_data;
    (void) nc;
    (void) ev_data;
    if (ev != MG_EV_POLL) {
        LOG(LL_DEBUG, ("sntp nc event %d", ev));
    }
}

struct mg_connection* sntp_bind() {
    struct mg_connection* nc = mg_connect(mgos_get_mgr(), str_connect_uri, sntp_connection_handler, NULL);
    if (nc == NULL) {
        LOG(LL_ERROR, ("unable to create SNTP UDP socket"));
    }
    return nc;
}

static void send_timer(void* arg) {
    (void)arg;
    char* ip = mgos_wifi_get_sta_ip();

    if(mgos_wifi_get_status() == MGOS_WIFI_IP_ACQUIRED && ip != NULL) {
        LOG(LL_INFO,("Try send"));
        mg_send(sntp_nc, str_sntp_request, sizeof(str_sntp_request));
    }

    free(ip);
}

enum mgos_app_init_result mgos_app_init(void) {
    // Create the UDP socket connection which is never destroyed
    sntp_nc = sntp_bind();

    // Initialize sntp request buffer, which never mutates
    memset(str_sntp_request, 0, sizeof(str_sntp_request));
    str_sntp_request[0] = '\x1b';

    // Start send timer
    mgos_set_timer(200, true, send_timer, NULL);

    return MGOS_APP_INIT_SUCCESS;
}
```
mos.yml
```
name: c_network
# arch: PLATFORM
version: 1.0
skeleton_version: 2017-03-17
sources:
  - src
filesystem:
build_vars:
  MGOS_ENABLE_DNS_SD: 0
  MGOS_ENABLE_UPDATER_POST: 1
  MGOS_ENABLE_SNTP: 0
  MGOS_ENABLE_MQTT: 0
  MGOS_ENABLE_ATCA: 0
  MGOS_ENABLE_ATCA_SERVICE: 0 
  MGOS_ENABLE_BITBANG: 0
  MGOS_ENABLE_RPC: 0
  MGOS_ENABLE_RPC_CHANNEL_HTTP: 0
  MGOS_ENABLE_RPC_CHANNEL_LOOPBACK: 0
  MGOS_ENABLE_RPC_CHANNEL_MQTT: 0
  MGOS_ENABLE_RPC_CHANNEL_UART: 0
  MGOS_ENABLE_DEBUG_UDP: 0
  APP_CONF_SCHEMA: src/conf_schema.yaml
```
conf_schema.yaml
```
[
  ["wifi.sta.enable", true],
  ["wifi.ap.enable", false],
  ["wifi.sta.ssid", "SSID"],
  ["wifi.sta.pass", "KEY"],
]
```
It seems the problem is in mg_send
```
void mg_send(struct mg_connection *nc, const void *buf, int len) {
  nc->last_io_time = (time_t) mg_time();
  if (nc->flags & MG_F_UDP) {
    nc->iface->vtable->udp_send(nc, buf, len);
  } else {
    nc->iface->vtable->tcp_send(nc, buf, len);
  }
}
```

```
dhcp client start...
mgos_wifi_on_change_ Wifi: connected
ip:192.168.63.140,mask:255.255.255.0,gw:192.168.63.1
mgos_wifi_on_change_ WiFi: ready, IP 192.168.63.140, DNS 192.168.63.1
send_timer           Try send

Exception 20 @ 0x00008000, vaddr 0x00008000
 A0: 0x40270ab4  A1: 0x3ffffe80  A2: 0x3ffef6cc  A3: 0x3ffee489
 A4: 0x00000030  A5: 0x00008000  A6: 0x7ff00000  A7: 0x7ff2af8d
 A8: 0x000003fe  A9: 0x58000000 A10: 0x00000403 A11: 0xfbe76c8b
A12: 0x3ffef6cc A13: 0x00000030 A14: 0x3ffee489 A15: 0x00000000

(exc SP: 0x3ffffce0)

--- BEGIN CORE DUMP ---
```

ceco, 2017-05-23:
Here is a traceback that describes the crash:

```
Reading symbols from /app.elf...done.
Remote debugging using 127.0.0.1:1234
Found core at 1857 - 179145
Mapping DRAM: 98304 @ 0x3ffe8000
Mapping IRAM: 32768 @ 0x40100000
Mapping IROM: 483328 @ 0x40208000
Mapping /mongoose-os/common/platforms/esp8266/rom/rom.bin at 0x40000000
Loaded core dump from last snippet in  /console.log
Unmapped addr 0x8000
0x00008000 in ?? ()
add symbol table from file "/mongoose-os/common/platforms/esp8266/rom/rom.elf" at
        .text_addr = 0x40000000
Reading symbols from /mongoose-os/common/platforms/esp8266/rom/rom.elf...(no debugging symbols found)...done.
Unmapped addr 0x8000
(gdb) bt
#0  0x00008000 in ?? ()
#1  0x40270ab4 in mg_send (nc=0x3ffef6cc, buf=buf@entry=0x3ffee489 <str_sntp_request>, len=len@entry=48) at mongoose/src/net.c:511
#2  0x4027d762 in send_timer (arg=<optimized out>) at src/main.c:37
#3  0x40273cd4 in mgos_timer_ev (nc=<optimized out>, ev=<optimized out>, ev_data=<optimized out>, user_data=0x3ffefeac) at /mongoose-os/fw/src/mgos_timers_mongoose.c:75
#4  0x4026b2e0 in mg_call (nc=nc@entry=0x3ffefebc, ev_handler=0x40273c24 <mgos_timer_ev>, ev_handler@entry=0x0, user_data=0x3ffefeac, ev=ev@entry=6, ev_data=ev_data@entry=0x3fffff10) at mongoose/src/net.c:91
#5  0x4026e305 in mg_if_timer (c=c@entry=0x3ffefebc, now=<optimized out>) at mongoose/src/net.c:115
#6  0x4026eb50 in mg_lwip_if_poll (iface=<optimized out>, timeout_ms=<optimized out>) at common/platforms/lwip/mg_lwip_ev_mgr.c:137
#7  0x4026f862 in mg_mgr_poll (m=m@entry=0x3ffef05c <s_mgr>, timeout_ms=timeout_ms@entry=0) at mongoose/src/net.c:285
#8  0x40275cba in mongoose_poll (ms=ms@entry=0) at /mongoose-os/fw/src/mgos_mongoose.c:55
#9  0x40275d3c in mongoose_poll_cb (arg=<optimized out>) at /mongoose-os/fw/platforms/esp8266/src/esp_main.c:78
#10 0x4027349f in mgos_lwip_task (e=<optimized out>) at /mongoose-os/fw/platforms/esp8266/src/esp_main.c:228
#11 0x40000f49 in ets_post ()

```

rojer, 2017-05-23:
can you post `debug.level=4` log?

ceco, 2017-05-25:
Actually we made this modification: https://gist.github.com/kzyapkov/48f5cbb77362658ade981bb7becbba16
which fixes the crash.

```
    switch (ev) {
        case MG_EV_CLOSE: {
            LOG(LL_INFO, ("%p Connection closed", nc));
            if (nc == sntp_nc) {
                sntp_nc = sntp_connect();
            }
            break;
        }
        case MG_EV_RECV: {
            LOG(LL_INFO, ("%d bytes", nc->recv_mbuf.len));
            mbuf_remove(&nc->recv_mbuf, nc->recv_mbuf.len);
        }
}
```

Could you confirm that this is correct use case ?

rojer, 2017-05-25:
having looked closer at this, i can now say that the behavior is valid and your fix is the right thing to do.
in your initial version, you create a connection and then use it forever to send things.
this would, in fact, work if the address were an IP address. but because address is a DNS name, it tries to resolve it first, fails, and at that point connection is closed and destroyed, after which using the same pointer leads to a crash.
when you properly handle MG_EV_CLOSE by re-creating the connection, the connection pointer is updated and old connection is no longer used, which is the right thing to do.
remember to always handle EV_CLOSE. no matter what the reason is, after EV_CLOSE connection is gone and the pointer must not be used again.

that said, i acknowledge that this behavior with DNS vs IP addresses is inconsistent. i'll keep that in mind. maybe instead of giving up on the connection and auto-closing it, we should retry DNS lookups until user explicitly closes the connection. we'll thing about it.

rojer, 2017-05-25:
so, i'm going to close it as WAI - for now, at least.

closed 2017-05-25 by rojer

kzyapkov, 2017-06-08:
This kind of works, but in some situations `MG_EV_CLOSE` is received immediately after connect, and we fall into a tight loop.