cesanta/mongoose-os #269: MQTT.pub() doesn't publish with QoS 2

DrSchottky, 2017-05-24:
I used the latest mos on ESP8266, and publishing messages with QoS 2 doesn't work.
MQTT.pub() returns 1, but the broker does not receive the message.
QoS 0 and 1 work fine.

bravokeyl, 2017-05-24:
@DrSchottky  Where are you publishing to? What does your debug logs say?

DrSchottky, 2017-05-24:
I used a personal broker and mosquitto/eclipse public brokers. The code is the button_mqtt.js example.
With QoS 1 log is
```
mgos_mqtt_pub        Publishing to mOS/topic1 @ 1 (36): [{"free_ram":39096,"total_ram":51912}]
Published: yes topic: mOS/topic1 message: {"free_ram":39096,"total_ram":51912} 
mgos_mqtt_ev         MQTT event: 204
```

With QoS 2

```
mgos_mqtt_pub        Publishing to mOS/topic1 @ 2 (36): [{"free_ram":39104,"total_ram":51912}]
Published: yes topic: mOS/topic1 message: {"free_ram":39104,"total_ram":51912} 
mgos_mqtt_ev         MQTT event: 205
```

P.S.: The WiFi connection is provided by a 3G AP

rojer, 2017-06-01:
yes, mongoose currently does not perform the second phase of QoS 2 publishing. event 205 is the PUBREC message which requires a PUBREL response. i'll look into that.

closed 2017-06-01 by rojer