cesanta/mongoose-os #270: "mos ui" ignores --port argument, without error

scotthernandez, 2017-05-26:
Starting mos with the port argument no longer works in the web-ui, nor does it generate an error and fail to start.

> mos --port ws://192.168.1.120/rpc ui

The port argument is still honored when using other commands, like "ls".
> mos --port ws://192.168.1.120/rpc ls
> file1
> file2

You can connect over the web socket via the connection dialog manually once the web-ui starts.

cpq, 2017-05-29:
Hmmmm yes.
What behaviour would you expect from the UI if the port is specified?

Tommystus, 2017-05-30:
Previous version works as as expected without having to specify the port again in ui dialog

scotthernandez, 2017-05-30:
Sounds like this has been changed and pushed already. Can anyone confirm?

If not, i should be able to in the next day or two.

cpq, 2017-05-31:
Not yet! Today or tomorrow.

We are about to push a big change as well - spliiting Mongoose OS into modular libraries. That will change the coding workflow from current "edit files on device directly" to what it should be "edit files in a local project then upload to device".

cpq, 2017-06-01:
The change is pushed. Please verify.

Two ways to specify port:
`mos --port XXXX`
or, export env variable
```bash
export MOS_PORT=XXXX
mos
```

Just in case, clear your browser cache.

closed 2017-06-01 by cpq