cesanta/mongoose-os #272: Unexpected operator error during installation.

nkolban, 2017-05-30:
I attempted to install mongoose-os on an Ubuntu system and received an error.  The complete log is shown next.   I haven't performed any diagnosis beyond reporting this error.

```
$ curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/sh
Downloading https://mongoose-os.com/downloads/mos/linux/mos ...
######################################################################## 100.0%
Installing into /home/kolban/.mos/bin/mos ...
/bin/sh: 44: test: 127: unexpected operator
SUCCESS: /home/kolban/.mos/bin/mos is installed.
Run '/home/kolban/.mos/bin/mos --help' to see all available commands.
Run '/home/kolban/.mos/bin/mos' without arguments to start a simplified Web UI installer.
```

cpq, 2017-05-31:
Thanks Neil.
That's the second time I see this error. What shell have you got on `/bin/sh`? Is it bash or some other shell?

cpq, 2017-05-31:
Thanks.

We've changed installation shell from `/bin/sh` to `/bin/bash`, that should fix the problem.
Please reopen if you still see an issue.

closed 2017-05-31 by cpq