cesanta/mongoose-os #275: ESP8266 AP<->STA Mode switch bug

iliapenev, 2017-05-31:
As known ESP8266 could work in AP+STA mode. 
I would like to use only AP or only STA mode. 
Switching modes are done with example below. 

There is a bug. Sometimes I get AP+STA mode, because switching to AP fails:
"mgos_wifi_setup_ap WiFi AP: Failed to set config"

Log:
https://gist.github.com/iliapenev/ca81d88141d45f973e16d57bc8b043f0

```

static void wifi_change_timer(void *arg) {
    (void) arg;
    struct sys_config *cfg = get_cfg();

    if(cfg->wifi.ap.enable) {
        LOG(LL_INFO, ("Restart to sta"));
        cfg->wifi.ap.enable = false; 
        cfg->wifi.sta.enable = true;
        mgos_wifi_setup_ap(&cfg->wifi.ap);
        mgos_wifi_setup_sta(&cfg->wifi.sta);
    } else if(cfg->wifi.sta.enable) {
        LOG(LL_INFO, ("Restart to ap"));
        cfg->wifi.sta.enable = false; 
        cfg->wifi.ap.enable = true;
        mgos_wifi_setup_sta(&cfg->wifi.sta);
        mgos_wifi_setup_ap(&cfg->wifi.ap);
    }
}
enum mgos_app_init_result mgos_app_init(void) {
    mgos_set_timer(30000 /* ms */, true /* repeat */, wifi_change_timer, NULL);
    return MGOS_APP_INIT_SUCCESS;
}
```

mos.yml

```
version: 1.0
skeleton_version: 2017-05-18
sources:
  - src
filesystem:
  - fs
config_schema:
  - ["debug.level", 4]
  - ["device.id", "test"]
  - ["wifi.sta.enable", false]
  - ["wifi.sta.ssid", "MySta"]
  - ["wifi.sta.pass", "MyStaPassword"]
  - ["wifi.ap.enable", true]
  - ["wifi.ap.ssid", "MyAP-??????"]
  - ["wifi.ap.pass", ""]
  - ["wifi.ap.ip", "192.168.150.1"]
  - ["wifi.ap.gw", "192.168.150.1"]
  - ["wifi.ap.dhcp_start", "192.168.150.2"]
  - ["wifi.ap.dhcp_end", "192.168.150.100"]
```

rojer, 2017-06-01:
hm. looks like SDK does not always transition from STA to disabled properly. will take a look.

rojer, 2017-06-01:
so, i don't see anything wrong with what we're doing.
and i actually was not able to reproduce the problem, even at a higher rate of switching (10 seconds between mode flips). what i see from your log is that sometimes device gets stuck in the AP mode and then when switching to STA we try to set AP+STA because get_opmode still reports AP.
i made the change above, which should change the behavior a bit - we will now try to set AP-only mode.
it may help. try it, let me know.

iliapenev, 2017-06-02:
Thanks! I will try the fix and will provide feedback.

iliapenev, 2017-06-02:
Seems that problem is gone. Over 3 hours I don't have combined wifi mode. 
Thanks

closed 2017-06-02 by iliapenev

andyjjones1, 2018-08-16:
Here is a "diff" between the log (at debug.level=3) between a STA+AP setup success and a fail (with success pointing to the left "<")
```
> pm open,type:2 0
142,143d143
< state: 5 -> 0 (0)
< rm 0
146,149d145
< mode : sta(60:01:94:09:c4:7c) + softAP(62:01:94:09:c4:7c)
< add if1
< dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
< bcn 100
151,156c147,148
< bcn 0
< del if1
< add if1
< dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
< bcn 100
< mgos_wifi_dev_ap_set WiFi AP IP: 192.168.4.1/255.255.255.0 gw 192.168.4.1, DHCP range 192.168.4.2 - 192.168.4.100
---
> mgos_wifi_dev_ap_set WiFi AP: Failed to set config
> wifi_setup_softap_se Andrew: SOFTAP setup fail.
```

andyjjones1, 2018-08-16:
NON-OS SDK says that ESP8266 Station mode must be on before calling wifi_station_set_config_current.  We see that MGOS calls mgos_wifi_add_mode(STATION_MODE)) just before wifi_station_set_config_current() but my theory is that sometimes on a lower level the mode change does not always happen quickly enough.