cesanta/mongoose-os #277: DHT11 under mongoose OS does not works ...

vgrebenschikov, 2017-06-04:
Exactly same wiring schema (trivial - only DHT11 sensor and ESP32 on devboard), same hardware ...

with plain ESP-IDF libraries and sample code from  Sam Johnston
it works fine, with Mongoose it fails on "Timeout waiting for start signal low pulse." 

on logic analyzer, signals looks similar (see screenshot) - on top - non-working capture from Mongoose and on bottom working with plain idf.

[image]

vgrebenschikov, 2017-06-04:
Also, is there right way to make all DEBUG_PRINT and DEBUG_PRINTLN works?

vgrebenschikov, 2017-06-04:
After applying the 

diff --git a/Adafruit_DHT/DHT.cpp b/Adafruit_DHT/DHT.cpp
index 78f48d8..4e97fcb 100644
--- a/Adafruit_DHT/DHT.cpp
+++ b/Adafruit_DHT/DHT.cpp
@@ -139,11 +139,11 @@ boolean DHT::read(bool force) {

   // Go into high impedence state to let pull-up raise data line level and
   // start the reading process.
+  pinMode(_pin, OUTPUT);
   digitalWrite(_pin, HIGH);
   delay(250);

   // First set data line low for 20 milliseconds.
-  pinMode(_pin, OUTPUT);
   digitalWrite(_pin, LOW);
   delay(20);

@@ -154,6 +154,7 @@ boolean DHT::read(bool force) {
     InterruptLock lock;

     // End the start signal by setting data line high for 40 microseconds.
+    pinMode(_pin, OUTPUT);
     digitalWrite(_pin, HIGH);
     delayMicroseconds(40);

cpq, 2017-07-09:
@ruslanvaliullin is that addresset already ?

ruslanvaliullin, 2017-07-11:
We need to go into high-Z state with pull-up. We should change to input mode, not to output mode and digitalWrite(_pin, HIGH). It's a known issue and will be addressed soon.

vgrebenschikov, 2017-07-14:
Time to time it works, what is strange for me.

Do you think that it will work on other GPOI pin?
What pins are ready for pull-up? 
(is it right direction to dig into: https://www.esp32.com/viewtopic.php?t=439 ?)

ruslanvaliullin, 2017-07-25:
@vgrebenschikov, the new DHT lib was implemented:

https://github.com/mongoose-os-libs/dht
usage examples:
https://github.com/mongoose-os-apps/example-dht-js
https://github.com/mongoose-os-apps/example-dht-c

Please check. Regards

vgrebenschikov, 2017-07-29:
Thanks, now it works!

closed 2017-07-29 by vgrebenschikov