cesanta/mongoose-os #280: WPS support.

bestpika, 2017-06-09:
Could Mongoose OS add WPS support for ESP8266 and ESP32.
Thank you.

cpq, 2017-06-15:
Do you mean https://en.wikipedia.org/wiki/Wi-Fi_Protected_Setup ?
If yes, could you elaborate on your project please?

bestpika, 2017-06-15:
Yes, I mean Wi-Fi Protected Setup.
I wish my device have less setting and easy to use, so user no need to enter any information, they just press WPS button and my device can auto connect to Internet.

scotthernandez, 2017-08-11:
As stated in the wikipedia page this feature is no longer suggested for security reasons. It is not secure and should not be used -- is is broken by design and allows an attack vector for brute-force attacks to be successful in milliseconds, or a few hours at the most. Just having it enabled in the access point (router) open up this attack vector, even if never actually used.

cpq, 2017-08-17:
Good point, Scott.
Closing this.

closed 2017-08-17 by cpq

bestpika, 2017-08-18:
If use push button method not pin mode, I think it still secure.

kunal-cloud-kickers, 2018-11-08:
Yes is it possible to push button on both end, from wifi router and esp?