cesanta/mongoose-os #281: Nmap cause reset on ESP8266

meinpilsbier, 2017-06-09:
Hi,
 if i used nmap to make a portscan on the device, the ESP8266 resets on a lot of ports.
How can i disable it ?
 
regards
/jörg

cpq, 2017-06-15:
What do you expect to see?

scotthernandez, 2017-06-17:
I would expect a port scan to not reset the esp8266/device. Is that what you want @meinpilsbier?

@meinpilsbier, can you provide the arguments you are using for the port scan so I could try to reproduce it?

Also, if reproducible, I would consider this a security issue if the device can be remotely reset by sending it a "random" packet, since that would be a denial-of-service (DOS) scenario.

meinpilsbier, 2017-06-19:
Hi cpq, as Scott commented,  it shoud not  reset.
I only use "nmap ip-address", to scan open ports. (OSX Sierra)

This is the serial dump

mgos_http_ev         0x3fff246c HTTP connection from 192.168.112.49:60671

Exception 28 @ 0x4027a8f2, vaddr 0x00000004
 A0: 0x4027aa7d  A1: 0x3ffffed0  A2: 0x3fff246c  A3: 0x00000000
 A4: 0x00000001  A5: 0x3fff19d4  A6: 0x3ffef8d4  A7: 0x0000041d
 A8: 0x000003fe  A9: 0x352220bc A10: 0x0000041d A11: 0x9a91105e
A12: 0x3fff19d4 A13: 0x3ffeee44 A14: 0x00000000 A15: 0x3fff246c

(exc SP: 0x3ffffd30)

--- BEGIN CORE DUMP ---
{"arch": "ESP8266",
"REGS": {"addr": 1073741152, "data": "
faonQND+/z9sJP8/AAAAAAEAAADUGf8/1Pj+Px0EAAD+AwAAvCAiNR0EAABeEJGa1Bn/P0Tu/j8AAAAAbCT/P/KoJ0AfAAAAAAAAAAAAAAAAAAAAMAAAAA==", "crc32": 3871232949},
"DRAM": {"addr": 1073643520, "data": "
GYAhQAAAAAAQgP4/AAAAAAIAAAD8gv4/ZIP+P8yD

I test it with ESP8266 and ESP32.
I only flash the ESP and set the WLAN data.

/jörg

cpq, 2017-07-15:
Ruslan, please take care about this!

ruslanvaliullin, 2017-08-16:
fixed

closed 2017-08-16 by ruslanvaliullin