cesanta/mongoose-os #283: api_mqtt.js no longer on ESP32 after june 9 update

DENightOne, 2017-06-10:
Updated this a few hours ago... api_mqtt.js is now missing from the ESP32 so cannot connect to MQTT
Error Log:
MJS error: failed to exec file "api_mqtt.js": failed to read file "api_mqtt.js"

fantawttw, 2017-06-12:
I'm also receiving the error.
Log Extract
```
esp32_mgos_init      mjs_base 1.0 (20170609-193726/???)
esp32_mgos_init      Mongoose OS Firmware 2017060919 (20170609-193726/???)
esp32_mgos_init      ESP-IDF v1.0-1373-g7e17ba7
esp32_mgos_init      Boot partition: app_0, Task ID: 0x3ffbc738, RAM: 224116 free
esp32_fs_mount       Mounting SPIFFS from fs_0 (131072 @ 0x190000)
esp32_fs_mount       Total: 113201, used: 76053, free: 37148
mgos_sys_config_init MAC: XXXXXXXXXXX
mgos_sys_config_init WDT: 30 seconds
mgos_wifi_set_mode   WiFi mode: STA
E (2942) wifi: esp_wifi_set_mode 774 wifi is not init
I (2942) wifi: wifi firmware version: 75d56a6
I (2952) wifi: config NVS flash: enabled
I (2952) wifi: config nano formating: disabled
[0;32mI (2962) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (2962) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (2972) wifi: Init dynamic tx buffer num: 32
I (2982) wifi: Init dynamic rx buffer num: 64
I (2982) wifi: wifi driver task: 3ffbd5e4, prio:23, stack:4096
I (2992) wifi: Init static rx buffer num: 10
I (3002) wifi: Init dynamic rx buffer num: 0
I (3002) wifi: Init rx ampdu len mblock:7
I (3012) wifi: Init lldesc rx ampdu entry mblock:4
I (3012) wifi: wifi power manager task: 0x3ffd0064 prio: 21 stack: 2560
E (3022) wifi: esp_wifi_connect 816 wifi not start
I (3032) wifi: wifi timer task: 3ffd10d0, prio:22, stack:3584
[0;32mI (3052) phy: phy_version: 354.0, b2dba61, May 16 2017, 21:41:06, 0, 0[0m
I (3062) wifi: mode : sta (XX:XX:XX:XX:XX:XX)
event_handler        event: 2
mgos_wifi_setup_sta  WiFi STA: Connecting to XXXXXXXX
mgos_i2c_create      I2C GPIO init ok (SDA: 32, SCL: 33)
mgos_sys_config_init HTTP server started on [80]
mg_rpc_channel_uart  0x3ffd1a08 UART0
  at init.js:3
MJS error: failed to exec file "api_mqtt.js": failed to read file "api_mqtt.js"
mg_rpc_channel_mqtt  0x3ffd238c esp32_10A13C/rpc/#
mgos_init            Init done, RAM: 257584 total, 183036 free, 178836 min free
```

DENightOne, 2017-06-13:
Seems they fixed this in yesterdays build (12 June)
Forum post here: https://forum.mongoose-os.com/discussion/1191/failed-to-read-file-api-mqtt-js

cpq, 2017-06-15:
Yeah, fixed, thanks for reporting

closed 2017-06-15 by cpq