cesanta/mongoose-os #284: ESP32 Analog IO

Galbi3000, 2017-06-11:
Any idea when this feature of the ESP32 will be supported in Mongoose OS? And if it already is then how do I use it as I can not find anything in the documentation about it?

cpq, 2017-06-15:
For now, ADC API is not yet FFI-ed to JS, we expect it to be done next week.

sctv, 2017-06-16:
Thanks
api_adc.js works nicely.
 Here is an example which reads an analog rain sensor when a button is pressed and publishes to the rain feed on a mqtt server. Sorry I can't get the insert code button to work with it.  

load('api_config.js');
load('api_gpio.js');
load('api_mqtt.js');
load('api_sys.js');
load('api_adc.js');

GPIO.set_button_handler(0, GPIO.PULL_UP, GPIO.INT_EDGE_NEG, 200, function() {
  let topic = Cfg.get('mqtt.user') + '/feeds/rain';
  let message = JSON.stringify(ADC.read(32));//reads pin 32 which is ADC1_CHANNEL_4 https://esp-idf.readthedocs.io/en/v2.0/api/peripherals/adc.html
  let ok = MQTT.pub(topic, message, 1);
  print('Published:', ok ? 'yes' : 'no', 'topic:', topic, 'message:', message);
}, null);

sctv, 2017-06-16:
@ruslanvaliullin @cpq I am new to this... is the above what you mean by the ADC API being FFI-ed to JS?
Thanks

ruslanvaliullin, 2017-06-17:
Yes that's it

cpq, 2017-07-15:
Closing, as done.

closed 2017-07-15 by cpq