cesanta/mongoose-os #285: Adjust ADC attenuation support

crashGoBoom, 2017-06-13:
Is there or can there be support for adjusting attenuation? Currently you have to modify `fw/platforms/esp32/src/esp32_hal.c` in order to adjust it. Seems to default to `ADC_ATTEN_11db` but the following would be nice to see added as options per pin via config or api.
```
ADC_ATTEN_0db
ADC_ATTEN_2_5db
ADC_ATTEN_6db
```
Thanks

cpq, 2017-06-15:
Please use ESP-IDF API for that - just include the header and call relevant function

closed 2017-06-15 by cpq

crashGoBoom, 2017-06-15:
So its possible to override the [mgos_adc_read](https://github.com/cesanta/mongoose-os/blob/master/fw/platforms/esp32/src/esp32_hal.c#L128
 ) in esp32_hal.c? Seems like it will still default to 11db if you try to configure it elsewhere. Do you have an example of how to do this? 
 Thanks! 
```
int mgos_adc_read(int pin) {
  static int pins[] = {36, 37, 38, 39, 32, 33, 34, 35};
  static adc1_channel_t channels[] = {
      ADC1_CHANNEL_0, ADC1_CHANNEL_1, ADC1_CHANNEL_2, ADC1_CHANNEL_3,
      ADC1_CHANNEL_4, ADC1_CHANNEL_5, ADC1_CHANNEL_6, ADC1_CHANNEL_7};
  int i;
  for (i = 0; i < ARRAY_SIZE(pins); i++) {
    if (pins[i] == pin) {
      adc1_config_width(ADC_WIDTH_12Bit);
      adc1_config_channel_atten(channels[i], ADC_ATTEN_11db);
      return 0x0FFF & adc1_get_voltage(channels[i]);
    }
  }
```

cpq, 2017-06-15:
No, not possible yet, let us expand `mgos_ads_*` API to allow that.
Expect next week.
If in hurry, just use ESP32 SDK API.

crashGoBoom, 2017-06-15:
@cpq no hurry! Next week sounds great! I'll be watching out for it. Thank you so much.

crashGoBoom, 2017-06-23:
Hi @cpq, was this still planned for this week?

cpq, 2017-06-23:
Yeah, we have finalised our ADC API:

https://github.com/cesanta/mongoose-os/blob/master/fw/src/mgos_adc.h#L22-L25

Setting ADC width, attenuation, is quite ESP32-specific and will not be part of cross-platform API.
Mongoose OS will set reasonable defaults:

https://github.com/cesanta/mongoose-os/blob/master/fw/platforms/esp32/src/esp32_adc.c

But if you want to override those, just use ESP32 SDK directly. I think it even might make sense to expose that to JS.

crashGoBoom, 2017-06-23:
Ok should something like this work in my own fs file?
`
let ESP_ADC = {
   let set6db: ffi('int adc1_config_channel_atten(int, ADC_ATTEN_6db))
}
`

Thanks for your help on this.

cpq, 2017-06-24:
```javascript
let MYADC = {
  cfg: ffi('int adc1_config_channel_atten(int, int)'),  // that function must be in the SDK
};

let db6 = 1234;  // whatever the numeric value for 6 db is
MYADC.cfg(pin, db6);
```