cesanta/mongoose-os #287: WebUI: Read mqtt logs when connected over web-socket/http

scotthernandez, 2017-06-14:
If the device is connected over WS/HTTP, and is configured with MQTT logging, then connect and read the logs from MQTT.

This could be generalized to -- read logs in whatever way the device is configured to log.

cpq, 2017-06-15:
If you're connected via http/websocket, the device is directly visible and therefore the UDP log is available. 

The setting is `debug.udp_log_addr="your_workstation_ip:1993"`
We just did not yet make it "magical".. The anticipated flow is: when mos tool connects to a device via websocket, it can ask "hey, device, what is the IP address you're seeing as mine?" then mos tool can `call Config.Set debug.udp_log_addr="that_ip:1993"` and bingo. you'll see the logs.

Note that UDP logs are way less expensive than MQTT. UDP loggin bypass everything, it just stuff things directly into the LWIP stack. Each message is stamped with sequence ID, so you can check in the log stream whether any message has been lost or not.

cpq, 2017-06-22:
That is done.
https://mongoose-os.com/docs/core_components/debug.html#debug-configuration-settings-quick-reference

closed 2017-06-22 by cpq