cesanta/mongoose-os #288: CPU Frequency

jocodeinc, 2017-06-15:
Could Mongoose OS add api in sys to retrieve and set CPU frequency for ESP8266 and ESP32? Something like sys.get_cpu_freq(); and sys.set_cpu_freq(mhz); where mhz can be 80, 160 or 240.

scotthernandez, 2017-06-17:
From what I've read, at least for the esp8266, there is no reason not to run at the top frequency since it doesn't effect power-usage in a noticeable way.

I'm curious, what is the motivation for this? What are the use-cases?

jocodeinc, 2017-06-19:
To increase speed for ESP32 which at default is currently running at 160. Would be nice to know what speed CPU are running at.

cpq, 2017-07-15:
I don't see much sense  mOS implementing cpu freq API
Closing - feel free to reopen if there are strong points to do that

closed 2017-07-15 by cpq