cesanta/mongoose-os #289: Use a consistent address scheme for ATCA crypto device I2C address specification

MartyMacGyver, 2017-06-15:
The code at https://github.com/cesanta/mongoose-os/blob/master/fw/src/mgos_atca.c#L201 implies an inconsistent contract:

	uint8_t addr = acfg->i2c_addr;
	  /*
	   * It's a bit unfortunate that Atmel requires address already shifted by 1.
	   * If user specifies address > 0x80, it must be already shifted since I2C bus
	   * addresses > 0x7f are invalid.
	   */
	  if (addr < 0x7f) addr <<= 1;

The Atmel specs for the crypto chips tend to refer to "8-bit" I2C addresses, but they're just standard 7-bit address with the R/W bit appended and set to zero. Therefore, there is no valid odd address.

Also, as this is, if a user configured their device to work at, say, 0x70 (Atmel) == 0x38 (7-bit), and they entered that 0x70 expecting it to work... it'll fail to work as expected.

Therefore, rather than arbitrarily adjust large addresses only, choose a contract: either use the standard 7-bit contract for the `acfg->i2c_addr` value or use the 8-bit Atmel one (noting that the real "addresses" will lop off the final bit (so something like 0xC1 --> 0xC0).

Also, specify *where the address get's entered* that the user is expected to conform to that contract if they enter a device address from the default one (e.g., if their device uses an Atmel's 8-bit address of 0xC0, and you expect the 7-bit contract, they need to pre-shift that right themselves and specify 0x60... if the 8-bit contract is what you stick with, note the loss of precision for user entries with the last bit set).

While it's normally preferable to stick to standard 7-bit I2C addressing, as this tends to be a OTP change the user who modifies their device will be used to the Atmel 8-bit format for their devices, thus sticking with the 8-bit address contract for this device's address spec is the most straightforward.

rojer, 2017-07-15:
@MartyMacGyver i agree with you, and this is what the code above aims to do. the library expects an 8-bit addr, and we allow it.if the address provided is < 127, i.e. 7.bit, we shift it by 1. so this should work naturally with default addresses. i don't see the problem here.

closed 2017-07-15 by rojer

MartyMacGyver, 2017-07-16:
The problem is that assuming an address with the 8th bit cleared is really a 7-bit address leads to ambiguity (disjoint behavior). If the user provides an address >= 0x80 it goes through as-is, but <0x80 it gets shifted - yet there are perfectly valid "8-bit" addresses below 0x80 (e.g. 0x70). Thus, if the user wants to enter 0x70 they really need to enter 0x38, but if they want to enter 0x80 they must enter 0x80.

I get that it's a convenience, but it's only a convenience if the user specifies an 8-bit address that's >= 0x7F. If they specify an 8-bit address < 0x7F, hopefully they know that what they specified will be treated as a 7-bit address. While addr > 0x80 == 8-bit is a reasonable assumption **for even values only**, it says nothing about the intent of values < 0x80, and only a look at the code will reveal that a conversion is happening.

This conversion table illustrates the ambiguity and side-effects of this algorithm - but if nothing else, it makes clear to the user what their input will be processed as, and what values will be allowed through despite being invalid 8-bit "addresses" (as odd 8-bit values toggle the R/W bit incorrectly).

	Given   shifted if < 0x7F
	    ....
	0x38    0x70
	0x39    0x72
	0x3A    0x74
	0x3B    0x76
	0x3C    0x78
	0x3D    0x7A
	0x3E    0x7C
	0x3F    0x7E
	0x40    0x80
	0x41    0x82
	    ....
	0x5E    0xBC
	0x5F    0xBE
	0x60    0xC0
	0x61    0xC2
	0x62    0xC4
	    ....
	0x7E    0xFC
	0x7F    0x7F  ---> outlier as-coded
	0x80    0x80
	0x81    0x81  (invalid)
	0x82    0x82
	    ....
	0xBC    0xBC
	0xBD    0xBD  (invalid)
	0xBE    0xBE
	0xBF    0xBF  (invalid)
	0xC0    0xC0
	0xC1    0xC1  (invalid)
	0xC2    0xC2
	0xC3    0xC3  (invalid)
	0xC4    0xC4
	    ....