cesanta/mongoose-os #292: init_archxx.js (init_esp8266.js , init_esp32.js) just like mos_xxarch.yml

bravokeyl, 2017-06-17:
Can we make arch based js file as the entry point if it's available else falling back to init.js?

scotthernandez, 2017-06-17:
I'm not sure this is a good idea. 

The yml files are combined whereas what you are suggesting would be more confusing since it would be an an "or", with non-obvious results not a combination or super-set. If you wanted to do what you suggest you should simply write that code into init.js to decide what js file to run based on the arch, or anything you want.

I am keen to make sure there is a good way to switch and reuse code based on the architecture and available libs, but that is another issue (which is probably not done yet).

cpq, 2017-07-15:
Yeah, agree with @scotthernandez

closed 2017-07-15 by cpq