cesanta/mongoose-os #294: Add support for persistence across deep-sleeps

scotthernandez, 2017-06-17:
For the esp8266 the rtc has storage which persists across deep-sleep and (re)start. I'm not surety  esp32 and other arch have support for similar features, but I hope so.

I'd like to see support for using this storage when doing deep-sleeps.

Particularly, I'd like to see something like this in JS:
```
let interestingData = {...};
// do stuff to make data interesting
...
deepSleep(..., interestingData, function(savedData) {
// on wakeup
interestingData = savedData;
// do some work, or set a timer, or whatever...
});
```

I'm not sure if init.js should be called when waking up from deep-sleep, or if the supplied function should simply be called first. But the idea is that all deepSleep/wake code is in one place which is clear and easy to understand.

mayson14, 2018-05-10:
Is this still not possible?  Trying to use RTC_DATA_ATTR

rojer, 2018-05-11:
yes, there is no mos-level support for deep sleep persistence yet. but you are welcome to use RTC_DATA_ATTR for now.

mayson14, 2018-05-11:
thank you for the repsonse.. do you have any mos examples of using RTC_DATA_ATTR?  i havent had much luck finding syntax samples..

DrBomb, 2018-05-11:
I found on google https://gist.github.com/igrr/54f7fbe0513ac14e1aea3fd7fbecfeab

rojer, 2018-05-11:
here's a snippet from an internal project that i wrote (with actual code cut out). but the boilerplate and comment are relevant.
put it into a file called `rtc_wake_stub_main.c` in the src dir (the `rtc_wake_stub_` prefix is important).
```c
#include "esp_sleep.h"
#include "rom/ets_sys.h"
#include "rom/uart.h"
#include "soc/timer_group_reg.h"
#include "soc/timer_group_struct.h"

/*
 * esp_wake_deep_sleep() is run by ROM very soon after wakeup from deep sleep.
 * Main RAM is available but contents have not been preserved so it's only good
 * for use as a scratch pad (be sure not to scrawl over the areas used by ROM).
 * Flash areas have no been mapped yet, but flash can be accessed via ROM funcs,
 * see flasher stub code for inspiration. Make sure you really use ROM though,
 * as IDF hides them pretty well and they are replaced with IRAM-based ones.
 * Those, of course, won't work. There are no exception handlers yet, so don't
 * expect any output on UART if the code crashes. TG0 WDT will restart the
 * device though in about 1 second though so it's not the end of the world.
 *
 * Note that .bss does not seem to be zeroed.
 *
 * Also note that at present data is placed into the RTC_SLOW segment,
 * which is, well, slow to access. Moving .rtc_data and .rtc_rodata to RTC_FAST
 * RAM will not only make it coexist better with ULP.
 * https://github.com/espressif/esp-idf/issues/1553 has been filed for that.
 *
 * NB: ESP32 has no hardware divider, so be sure not to use integer division.
 *
 * See http://esp-idf.readthedocs.io/en/latest/api-guides/deep-sleep-stub.html
 * for more details.
 */

uint32_t g_rtc_ram_data_flag;

void esp_wake_deep_sleep(void) {
  esp_default_wake_deep_sleep();
  /* Do stuff here */
  g_rtc_ram_data_flag = 1;
  uart_tx_one_char('!');
  uart_tx_one_char('\n');
}
```

it should print an exclamation mark at boot when coming out of deep sleep.

mayson14, 2018-05-11:
ah, ok.. the project im working on started from a template using the api_esp32.js javascript wrapper... i may have to convert my project to C.