cesanta/mongoose-os #297: Cloud build returns error message, but actually works

MrSurly, 2017-06-20:
Performing an `mos init --arch esp32` followed by `mos build` displays `Error: build failed`

It's actually successful in that you do get build artifacts that are useful for `mos flash`

```
$ mos init --arch esp32
Connecting to http://mongoose.cloud, user  ...
Downloading project skeleton...
Unpacking...
Setting arch "esp32"...
$ mos build
===
WARNING: Setting build variable "APP_CONF_SCHEMA" in "mos.yml" is deprecated, use "config_schema" property instead.
===
Connecting to http://mongoose.cloud, user test
Uploading sources (1566 bytes)
The flag --repo is not given, going to use mongoose-os repository
Sources: [src/*.c src/*.cpp]
Building...
===
WARNING: Setting build variable "APP_CONF_SCHEMA" in "mos.yml" is deprecated, use "config_schema" property instead.
===
Docker arguments: run --rm -i -v /tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124:/app -v /tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/modules/mongoose-os:/mongoose-os -v /tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/modules/mongoose-os:/tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/modules/mongoose-os -v /tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/src:/tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/src --user 0:0 docker.cesanta.com/esp32-build:2.0-r7 /bin/bash -c nice make '-j' '16' '-C' '/app' '-f' '/mongoose-os/fw/platforms/esp32/Makefile.build' 'APP=fw' 'APP_VERSION=1.0' 'FS_STAGING_DIR=build/fs' 'APP_SOURCES=src/*.c src/*.cpp' 'PLATFORM=esp32' 'GEN_DIR=/tmp/fwbuild-volumes/users/test/fw/build_requests/build_req_572187124/build/gen' 'MGOS_PATH=/mongoose-os' 'APP_CFLAGS=' 'BUILD_DIR=build/objs' 'APP_FS_FILES=' 'APP_CXXFLAGS=' 'APP_CONF_SCHEMA=src/conf_schema.yaml' 'FFI_SYMBOLS=' 'FW_DIR=build/fw'
make: Entering directory '/app'
GENCONFIG

<< Removed a bunch of compilation messages >> 

  MKFS  build/fs (131072) -> /app/build/objs/fw_temp/fs.img
     Adding sys_ro_vars_schema.json: 332
     Adding sys_config_schema.json: 4007
     Adding conf0.json: 1458
     Adding index.html: 825
     Adding ca.pem: 7845
     Image stats: size=131072, space: total=113201, used=16315, free=96886
  GEN    /app/build/objs/fw_temp/manifest.json
  ZIP    /app/build/fw/fw-esp32-last.zip
     Adding /app/build/objs/fw_temp/bootloader.bin
     Adding /app/build/objs/fw_temp/fs.img
     Adding /app/build/objs/fw_temp/fw.bin
     Adding /app/build/objs/fw_temp/partitions_mgos.bin
  Built fw/esp32 version 1.0 (20170620-202529/???)
To flash all build output, run 'make flash' or:
python /opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 1500000 --before default_reset --after hard_reset write_flash -z --flash_mode qio --flash_freq 40m --flash_size detect 0x1000 /app/build/objs/bootloader/bootloader.bin 0x10000 /app/build/objs/fw.bin 0x8000 /app/build/objs/partitions_mgos.bin
make: Leaving directory '/app'
Error: open build/objs/fw.elf: no such file or directory
Error: build failed
```

cpq, 2017-06-21:
@MrSurly thanks for reporting, fixing this.

dimonomid, 2017-06-21:
@MrSurly honestly, I couldn't reproduce the issue, for me `mos init --arch esp32` followed by `mos build` worked just fine.

Anyway, we've updated `mos init` to use the [empty app](https://github.com/mongoose-os-libs/empty) as a source, that's a long overdue.

Could you please check if it works for you now? If not, make sure you're running latest `mos` (to update, run `mos update`)

MrSurly, 2017-06-21:
This is on Ubuntu 16.10.

Perfomed `mos update`
```
$ mos --version
The Mongoose OS command line tool
Version: 2017062120
Build ID: 20170621-202027/???

$ mos init 
Downloading empty app...
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x63 pc=0xc507bb46]

runtime stack:
runtime.throw(0x8686817, 0x2a)
	/usr/local/go/src/runtime/panic.go:596 +0x7c
runtime.sigpanic()
	/usr/local/go/src/runtime/signal_unix.go:274 +0x28c

goroutine 40 [syscall, locked to thread]:
runtime.cgocall(0x84d2520, 0xd7b246e4, 0x8685c17)
	/usr/local/go/src/runtime/cgocall.go:131 +0xe1 fp=0xd7b246c8 sp=0xd7b246ac
net._C2func_getaddrinfo(0xc5c00510, 0x0, 0xd7b0cfe0, 0xd7af1f18, 0x0, 0x0, 0x0)
	net/_obj/_cgo_gotypes.go:85 +0x57 fp=0xd7b246e4 sp=0xd7b246c8
net.cgoLookupIPCNAME.func2(0xc5c00510, 0x0, 0xd7b0cfe0, 0xd7af1f18, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/cgo_unix.go:151 +0x133 fp=0xd7b24710 sp=0xd7b246e4
net.cgoLookupIPCNAME(0xd7b0cf00, 0x13, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/cgo_unix.go:151 +0x173 fp=0xd7b24790 sp=0xd7b24710
net.cgoIPLookup(0xd7b02f80, 0xd7b0cf00, 0x13)
	/usr/local/go/src/net/cgo_unix.go:203 +0x2d fp=0xd7b247e0 sp=0xd7b24790
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1629 +0x1 fp=0xd7b247e4 sp=0xd7b247e0
created by net.cgoLookupIP
	/usr/local/go/src/net/cgo_unix.go:213 +0x98

goroutine 1 [select]:
net/http.(*Transport).getConn(0x894cf40, 0xd7b12860, 0x0, 0xd7b02b00, 0x5, 0xd7b0cf00, 0x17, 0x0, 0x0, 0x17)
	/usr/local/go/src/net/http/transport.go:913 +0x90c
net/http.(*Transport).RoundTrip(0x894cf40, 0xd77d3d00, 0x894cf40, 0x0, 0x0)
	/usr/local/go/src/net/http/transport.go:378 +0x459
net/http.send(0xd77d3d00, 0x8952780, 0x894cf40, 0x0, 0x0, 0x0, 0x0, 0x0, 0x8292091, 0xd7b10758, ...)
	/usr/local/go/src/net/http/client.go:249 +0x137
net/http.(*Client).send(0x8c0bde0, 0xd77d3d00, 0x0, 0x0, 0x0, 0x0, 0x7f, 0x0, 0x8952320, 0xd7774030)
	/usr/local/go/src/net/http/client.go:173 +0xc9
net/http.(*Client).Do(0x8c0bde0, 0xd785c400, 0xd78e2640, 0x3c, 0x0)
	/usr/local/go/src/net/http/client.go:595 +0x1fa
net/http.(*Client).Get(0x8c0bde0, 0xd78e2640, 0x3c, 0x3c, 0xd78e2640, 0xd785c300)
	/usr/local/go/src/net/http/client.go:393 +0x83
net/http.Get(0xd78e2640, 0x3c, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/http/client.go:367 +0x33
main.initFW(0x8956440, 0xd77720cc, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/initfw.go:37 +0x126
main.run(0xd77f3260, 0x8956440, 0xd77720cc, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/main.go:125 +0x6c
main.main()
	/go/src/cesanta.com/mos/main.go:211 +0x418

goroutine 17 [syscall, locked to thread]:
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1629 +0x1

goroutine 5 [select]:
cesanta.com/vendor/github.com/golang/glog.(*loggingT).flushDaemon(0x8c0c1e0)
	/go/src/cesanta.com/vendor/github.com/golang/glog/glog.go:927 +0x11a
created by cesanta.com/vendor/github.com/golang/glog.init.1
	/go/src/cesanta.com/vendor/github.com/golang/glog/glog.go:441 +0x250

goroutine 37 [select]:
net/http.(*persistConn).writeLoop(0xd78cc1e0)
[... 44 lines cut ...]
	/usr/local/go/src/bufio/bufio.go:97 +0xd5
bufio.(*Reader).Peek(0xd7b10690, 0x1, 0xd779edad, 0x1, 0x0, 0xd7b02a00, 0x0)
	/usr/local/go/src/bufio/bufio.go:129 +0x51
net/http.(*persistConn).readLoop(0xd78cc1e0)
	/usr/local/go/src/net/http/transport.go:1474 +0x185
created by net/http.(*Transport).dialConn
	/usr/local/go/src/net/http/transport.go:1117 +0x88b

goroutine 39 [select]:
net.cgoLookupIP(0x8956480, 0xd7b02f00, 0xd7b0cf00, 0x13, 0x0, 0xd7848620, 0xd78fc028, 0x8951148, 0xd78ee0b0, 0xd784bb00)
	/usr/local/go/src/net/cgo_unix.go:214 +0x252
net.(*Resolver).lookupIP(0x8c21df2, 0x8956480, 0xd7b02f00, 0xd7b0cf00, 0x13, 0x0, 0x0, 0x0, 0x8c21f44, 0x0)
	/usr/local/go/src/net/lookup_unix.go:69 +0xc8
net.(*Resolver).(net.lookupIP)-fm(0x8956480, 0xd7b02f00, 0xd7b0cf00, 0x13, 0xd78fc000, 0x894bd20, 0x8957480, 0xd78562a0, 0x8957480)
	/usr/local/go/src/net/lookup.go:156 +0x40
net.glob..func11(0x8956480, 0xd7b02f00, 0xd7af1f08, 0xd7b0cf00, 0x13, 0x0, 0x0, 0x303, 0xd784c378, 0x0)
	/usr/local/go/src/net/hook.go:19 +0x3c
net.(*Resolver).LookupIPAddr.func1(0xd78fa0a0, 0x8295d3e, 0x894cf40, 0x8956440)
	/usr/local/go/src/net/lookup.go:162 +0x45
internal/singleflight.(*Group).doCall(0x8c0bb10, 0xd7b108a0, 0xd7b0cf00, 0x13, 0xd7b0cf80)
	/usr/local/go/src/internal/singleflight/singleflight.go:93 +0x21
created by internal/singleflight.(*Group).DoChan
	/usr/local/go/src/internal/singleflight/singleflight.go:86 +0x26d

```

dimonomid, 2017-06-22:
Just to let you know: so far we failed to reproduce it, probably we'll have to install Ubuntu 16.10 and test it there (so far, tested on Ubuntu 16.04 and Linux Mint 17.3, works fine).

Will let you know once we have something.

Maybe it's some sort of binary incompatibility, really hard to say. As a temporary measure, you can try to build `mos` tool and see if it helps; instructions are there https://github.com/cesanta/mongoose-os/tree/master/mos

MrSurly, 2017-06-22:
I ran this 26 times: `rm -Rf *; mos init --arch esp32`.  Of those 26 times, it worked 6 times (23%).  Given that it's failing in `net.cgoLookupIP`

Quick web search seems to [indicate](https://github.com/go-gitea/gitea/issues/1408) there may be problems with the cgo implementation of LookupIp.

The build instructions don't work for me:

* Installed go 1.8.3
* cloned mongoose-os into `$GOPATH/src/cesanta.com`
* installed `govendor`
* performed `govendor sync` in `$GOPATH/src/cesanta.com`
* performed `go build` in `$GOPATH/src/cesanta.com`:
    * `can't load package: package cesanta.com: no buildable Go source files in ~/workspaces/go/src/cesanta.com`
* tried `go build` in `$GOPATH/src/cesanta.com/mos`
    * `cannot find package "github.com/cesanta/ubjson"`
    * `cannot find package "github.com/fatih/color"`
    * `cannot find package "golang.org/x/net/trace"`
    * `cannot find package "golang.org/x/net/websocket"`
    * `cannot find package "gopkg.in/yaml.v2"`

Fixed this by:
`go get github.com/cesanta/ubjson`
`go get github.com/fatih/color`
`go get golang.org/x/net/trace`
`go get gopkg.in/yaml.v2`

tried `go build` in `$GOPATH/src/cesanta.com/mos`:

```# cesanta.com/mos
./flagutils.go:94: undefined: BuildId
./flagutils.go:102: undefined: BuildId
./main.go:190: undefined: Version
./main.go:190: undefined: BuildId
./ui.go:326: undefined: BuildId
./update.go:53: undefined: BuildId
./update.go:56: undefined: BuildId
```

Seeing as I know nothing of golang, this is where I gave up trying to build; google searches were not fruitful.

dimonomid, 2017-06-22:
@MrSurly thanks for tracking that down that far. Yes apparently our build instructions are obsolete, sorry about that. I'll update them soon.

Please run `go generate` before `go build`, and then build will work.

MrSurly, 2017-06-22:
I think it's probably been a long time since you've set up your dev environment on a fresh machine ...

Also needed:
* pygit
* go-bindata
    * vet (tries to install vet, which is now part of go, so it fails)
    * errcheck
    * golint
* go-bindata-assetfs

Another trip down the rabbit hole:

```
$ pwd 
$GOPATH/src/cesanta.com/mos

$ go generate
Traceback (most recent call last):
  File "../common/tools/fw_meta.py", line 48, in <module>
    import git
ImportError: No module named git
main.go:1: running "../common/tools/fw_meta.py": exit status 1

#
### Install pygit
#

$ sudo pip install pygit
Collecting pygit
  Downloading pygit-0.1.tar.gz
Installing collected packages: pygit
  Running setup.py install for pygit ... done
Successfully installed pygit-0.1

$ go generate
main.go:2: running "go-bindata": exec: "go-bindata": executable file not found in $PATH

#
### install go-bindata
#

$ go get github.com/jteeuwen/go-bindata
$ cd ../../github.com/jteeuwen/go-bindata/go-bindata/
$make
make -C testdata
make[1]: Entering directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
go get github.com/kisielk/errcheck
go get golang.org/x/tools/cmd/vet
package golang.org/x/tools/cmd/vet: cannot find package "golang.org/x/tools/cmd/vet" in any of:
	/usr/local/go/src/golang.org/x/tools/cmd/vet (from $GOROOT)
	/home/epoulsen/workspaces/go/src/golang.org/x/tools/cmd/vet (from $GOPATH)
Makefile:34: recipe for target 'vet' failed
make[1]: *** [vet] Error 1
make[1]: Leaving directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2

$ vim testdata/Makefile # Modifed "vet" target to be NOOP
$ make 
make -C testdata
make[1]: Entering directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
go get github.com/kisielk/errcheck
#go get golang.org/x/tools/cmd/vet
echo NOVET
NOVET
go get github.com/golang/lint/golint
go install ../...
/home/epoulsen/workspaces/go/bin/go-bindata -nomemcopy             -o out/compress-nomemcopy.go in/...
errcheck out/compress-nomemcopy.go
make[1]: errcheck: Command not found
Makefile:7: recipe for target 'out/compress-nomemcopy.checked' failed
make[1]: *** [out/compress-nomemcopy.checked] Error 127
make[1]: Leaving directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2

#
### install errcheck
#

$ go get github.com/kisielk/errcheck
$ cd ../../kisielk/errcheck
$ go build
$ cd -
$ cp ../../kisielk/errcheck/errcheck .
$ export PATH=$PATH:.:..
$ make
make -C testdata
make[1]: Entering directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
go get github.com/kisielk/errcheck
#go get golang.org/x/tools/cmd/vet
echo NOVET
NOVET
go get github.com/golang/lint/golint
errcheck out/compress-nomemcopy.go
go tool vet --all out/compress-nomemcopy.go
go tool vet --shadow out/compress-nomemcopy.go
golint out/compress-nomemcopy.go
make[1]: golint: Command not found
Makefile:7: recipe for target 'out/compress-nomemcopy.checked' failed
make[1]: *** [out/compress-nomemcopy.checked] Error 127
make[1]: Leaving directory '/home/epoulsen/workspaces/go/src/github.com/jteeuwen/go-bindata/testdata'
Makefile:2: recipe for target 'all' failed
make: *** [all] Error 2

#
### install golint
#

$ go get github.com/golang/lint
$ cd ../../golang/lint/golint/
$ go build
$ cd -
$ cp ../../golang/lint/golint/golint .
$ make
<stuff>
## Success!

$ cd go-bindata
$ go build

$ cd ../../../../cesanta.com/mos
$ cp ../../github.com/jteeuwen/go-bindata/go-bindata/go-bindata .
$ go generate
go generate
main.go:3: running "go-bindata-assetfs": exec: "go-bindata-assetfs": executable file not found in $PATH

#
### install go-bindata-assetfs
#

$ go get github.com/elazarl/go-bindata-assetfs/...
$ cd ../../github.com/elazarl/go-bindata-assetfs/go-bindata-assetfs
$ go build
$ cd -
$ cp ../../github.com/elazarl/go-bindata-assetfs/go-bindata-assetfs/go-bindata-assetfs .

# 
###  Finally, it builds
#

$ go generate
$ go build

#
### Lets try running 
# 

$ ./mos
mos flag redefined: log_dir
panic: mos flag redefined: log_dir

goroutine 1 [running]:
flag.(*FlagSet).Var(0xc4200800c0, 0xffbf00, 0xc420084f80, 0xadfbdc, 0x7, 0xafd9c7, 0x2f)
	/usr/local/go/src/flag/flag.go:793 +0x420
flag.(*FlagSet).StringVar(0xc4200800c0, 0xc420084f80, 0xadfbdc, 0x7, 0x0, 0x0, 0xafd9c7, 0x2f)
	/usr/local/go/src/flag/flag.go:696 +0x8b
flag.(*FlagSet).String(0xc4200800c0, 0xadfbdc, 0x7, 0x0, 0x0, 0xafd9c7, 0x2f, 0xc420084f70)
	/usr/local/go/src/flag/flag.go:709 +0x90
flag.String(0xadfbdc, 0x7, 0x0, 0x0, 0xafd9c7, 0x2f, 0xc42008a400)
	/usr/local/go/src/flag/flag.go:716 +0x69
cesanta.com/vendor/github.com/golang/glog.init()
	/home/epoulsen/workspaces/go/src/cesanta.com/vendor/github.com/golang/glog/glog_file.go:41 +0x14a
cesanta.com/common/go/mgrpc/codec.init()
	/home/epoulsen/workspaces/go/src/cesanta.com/common/go/mgrpc/codec/websocket.go:127 +0x87
cesanta.com/common/go/mgrpc.init()
	/home/epoulsen/workspaces/go/src/cesanta.com/common/go/mgrpc/transport_string.go:17 +0x61
cesanta.com/fw/defs/atca.init()
	/home/epoulsen/workspaces/go/src/cesanta.com/fw/defs/atca/gen.go:5 +0x58
main.init()
	/home/epoulsen/workspaces/go/src/cesanta.com/mos/wifi.go:26 +0xa3
```

dimonomid, 2017-06-23:
First of all, thanks for digging into the rabbit hole. Yes, we clearly might have done a better job documenting how to build mos. Sorry! :(

I also have to say that you started from providing me with a runtime mos issue which I can't reproduce, and now you have a build issue which I can't reproduce! My the only guess is that you might have mixed some vendored packages while trying to make this thing work. I also added missing packages to vendor.json, so that `go get github.com/cesanta/ubjson` and others are not needed. So, could you please do a bit of update and cleanup (assuming you don't have any valuable Go sources under `/home/epoulsen/workspaces/go/src` other than cesanta.com; if you do, then you'll have to be more selective when removing stuff) :

```
# pull latest revision of mongoose-os
cd /home/epoulsen/workspaces/go/src/cesanta.com
git pull

# remove all vendored packages
rm -r $(find vendor -maxdepth 1 -mindepth 1 -type d)

# remove all Go sources other than cesanta.com (assuming you don't have anything valuable there)
rm -r $(find /home/epoulsen/workspaces/go/src -maxdepth 1 -mindepth 1 -type d -not -name cesanta.com)

# fetch all required vendored package under /home/epoulsen/workspaces/go/src/cesanta.com/vendor
govendor sync

# once more, try to build mos
cd mos
go generate
go build
```

Please let me know if it helps

MrSurly, 2017-06-24:
@dimonomid 
Thank you for the updated instructions.  I _might_ have time to try this tomorrow on a fresh machine.  Otherwise, I'm out-of-town for a week, and I can try it after I return.

MrSurly, 2017-07-03:
@dimonomid 

### Building MOS from scratch:

A few issues, installing on a clean vagrant box running 16.10:
```
vagrant@vagrant:~/workspaces/go/src/cesanta.com/mos$ go generate
main.go:2: running "go-bindata": exec: "go-bindata": executable file not found in $PATH
```
I did follow the installation instructions (that state it will install the CLI executable) from the go-bindata repo -- didn't help.

### Running `mos init --arch esp32` on a clean 16.10 box:

Using vagrant + Virtualbox and a fresh boxcutter/ubuntu1610 image

```
vagrant@vagrant:~/workspaces/mongooseos-test$ mos init --arch esp32
Downloading empty app...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x84aa42e]

goroutine 1 [running]:
main.initFW(0x893d7e0, 0xd771408c, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/initfw.go:38 +0x12e
main.run(0xd7808480, 0x893d7e0, 0xd771408c, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/main.go:125 +0x6c
main.main()
	/go/src/cesanta.com/mos/main.go:215 +0x470
vagrant@vagrant:~/workspaces/mongooseos-test$
```

MrSurly, 2017-07-03:
### Running `mos init --arch esp32` on a clean 16.04 box:

tl;dr: Looks like original problem, but does _not_ produce any artifacts.

```
vagrant init ubuntu/xenial64
A `Vagrantfile` has been placed in this directory. You are now
ready to `vagrant up` your first virtual environment! Please read
the comments in the Vagrantfile as well as documentation on
`vagrantup.com` for more information on using Vagrant.
epoulsen@carambola ~/workspaces/vagrant/ubuntu-16.04 $ vagrant up
< much removed> 
epoulsen@carambola ~/workspaces/vagrant/ubuntu-16.04 $ vagrant ssh
Welcome to Ubuntu 16.04.2 LTS (GNU/Linux 4.4.0-83-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  Get cloud support with Ubuntu Advantage Cloud Guest:
    http://www.ubuntu.com/business/services/cloud

0 packages can be updated.
0 updates are security updates.

ubuntu@ubuntu-xenial:~$ curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/bash
Downloading https://mongoose-os.com/downloads/mos/linux/mos ...
######################################################################## 100.0%
Installing into /home/ubuntu/.mos/bin/mos ...
Adding /home/ubuntu/.mos/bin to your PATH in /home/ubuntu/.bashrc
SUCCESS: /home/ubuntu/.mos/bin/mos is installed.
Run '/home/ubuntu/.mos/bin/mos --help' to see all available commands.
Run '/home/ubuntu/.mos/bin/mos' without arguments to start a simplified Web UI installer.
ubuntu@ubuntu-xenial:~/workspaces/mos-test$ . ~/.bashrc 
ubuntu@ubuntu-xenial:~/workspaces/mos-test$ mos init --arch esp32
Downloading empty app...
Unpacking...
Setting arch "esp32"...
ubuntu@ubuntu-xenial:~/workspaces/mos-test$ mos build

<  removed >

CC mgos_service_config.o
  MKDIR /app/build/objs/fw_temp
/tmp/fwbuild-volumes/users/test/mos-test/build_requests/build_req_311133621/local_libs/rpc-service-config/src/mgos_service_config.c:7:22: fatal error: mgos_rpc.h: No such file or directory
compilation terminated.
/mongoose-os/fw/platforms/esp32/src/component.mk:129: recipe for target 'mgos_service_config.o' failed
make[1]: *** [mgos_service_config.o] Error 1
make[1]: Leaving directory '/app/build/objs/src'
/opt/Espressif/esp-idf/make/project.mk:409: recipe for target 'src-build' failed

< removed> 

AR libbt.a
make[1]: Leaving directory '/app/build/objs/bt'
make: Leaving directory '/app'
Error: exit status 2
Error: build failed
ubuntu@ubuntu-xenial:~/workspaces/mos-test$ mos update
Up to date.
ubuntu@ubuntu-xenial:~/workspaces/mos-test$ 
```

MrSurly, 2017-07-12:
Bump

Can you reproduce these issues?

dimonomid, 2017-07-13:
Yeah sorry for the delay.

> `main.go:2: running "go-bindata": exec: "go-bindata": executable file not found in $PATH`

`go get` installs binaries into `$GOPATH/bin`, so please do this: (you might want to add it to your `.bashrc` or equivalent file for your shell)
```bash
export PATH="$PATH:$GOPATH/bin"
```

W.r.t. the issues with pre-built binary, no, sorry, can't reproduce yet.

MrSurly, 2017-07-13:
> W.r.t. the issues with pre-built binary, no, sorry, can't reproduce yet.

Not even in a [vagrant](https://www.vagrantup.com/) instance?

dimonomid, 2017-07-15:
Yes, not even in a vagrant instance; I replayed the same scenario you posted

```
➜  ~/mydata/tmp/180/1 vagrant up                  
Bringing machine 'default' up with 'virtualbox' provider...
==> default: Box 'ubuntu/xenial64' could not be found. Attempting to find and install...
    default: Box Provider: virtualbox
    default: Box Version: >= 0
==> default: Loading metadata for box 'ubuntu/xenial64'
    default: URL: https://vagrantcloud.com/ubuntu/xenial64
==> default: Adding box 'ubuntu/xenial64' (v20170713.0.0) for provider: virtualbox
    default: Downloading: https://app.vagrantup.com/ubuntu/boxes/xenial64/versions/20170713.0.0/providers/virtualbox.box
==> default: Successfully added box 'ubuntu/xenial64' (v20170713.0.0) for 'virtualbox'!
==> default: Importing base box 'ubuntu/xenial64'...
==> default: Matching MAC address for NAT networking...
==> default: Checking if box 'ubuntu/xenial64' is up to date...
==> default: Setting the name of the VM: 1_default_1500144052966_65475
==> default: Clearing any previously set network interfaces...
==> default: Preparing network interfaces based on configuration...
    default: Adapter 1: nat
==> default: Forwarding ports...
    default: 22 (guest) => 2222 (host) (adapter 1)
==> default: Running 'pre-boot' VM customizations...
==> default: Booting VM...
==> default: Waiting for machine to boot. This may take a few minutes...
    default: SSH address: 127.0.0.1:2222
    default: SSH username: ubuntu
    default: SSH auth method: password
    default: 
    default: Inserting generated public key within guest...
    default: Removing insecure key from the guest if it's present...
    default: Key inserted! Disconnecting and reconnecting using new SSH key...
==> default: Machine booted and ready!
==> default: Checking for guest additions in VM...
==> default: Mounting shared folders...
    default: /vagrant => /home/dimon/mydata/tmp/180/1
➜  ~/mydata/tmp/180/1 vagrant ssh
Welcome to Ubuntu 16.04.2 LTS (GNU/Linux 4.4.0-83-generic x86_64)

 * Documentation:  https://help.ubuntu.com
 * Management:     https://landscape.canonical.com
 * Support:        https://ubuntu.com/advantage

  Get cloud support with Ubuntu Advantage Cloud Guest:
    http://www.ubuntu.com/business/services/cloud

0 packages can be updated.
0 updates are security updates.

ubuntu@ubuntu-xenial:~$ curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/bash
Downloading https://mongoose-os.com/downloads/mos-release/linux/mos ...
######################################################################## 100.0%
Installing into /home/ubuntu/.mos/bin/mos ...
Adding /home/ubuntu/.mos/bin to your PATH in /home/ubuntu/.bashrc
SUCCESS: /home/ubuntu/.mos/bin/mos is installed.
Run '/home/ubuntu/.mos/bin/mos --help' to see all available commands.
Run '/home/ubuntu/.mos/bin/mos' without arguments to start a simplified Web UI installer.
ubuntu@ubuntu-xenial:~$ mkdir 1
ubuntu@ubuntu-xenial:~$ cd 1
ubuntu@ubuntu-xenial:~/1$ . ~/.bashrc
ubuntu@ubuntu-xenial:~/1$ mos init --arch esp32
Downloading empty app...
Unpacking...
Setting arch "esp32"...
ubuntu@ubuntu-xenial:~/1$ mos build
Connecting to https://mongoose.cloud, user test
Uploading sources (49412 bytes)
Firmware saved to build/fw.zip
ubuntu@ubuntu-xenial:~/1$ 
```

dimonomid, 2017-07-15:
@MrSurly, however, I think you might have some "dirty" libs in your `~/.mos/libs`, so they are used as they are, and it might cause conflicts with the latest core code.

To confirm that, could you please run `mos build --verbose`, and post the log here?

Also, to try to fix it, you can remove/rename `~/.mos/libs` and try build again. Btw, also please don't forget to install `mos` again.

MrSurly, 2017-07-17:
> Yes, not even in a vagrant instance; I replayed the same scenario you posted

The issue you tested is indeed fixed now, but there are others.  Let me summarize the various issues from this thread: 

- [x] (1) issue with remote builds -- fixed (this is the original issue)
- [ ] (2) building of MOS from scratch (I'll try again)
- [ ] (3) 16.10 (yakkety in vagrant) issue with MOS crashing locally, when issuing `mos init --arch esp32` -- still an issue.
- [ ] (4) 16.10 (yakkety, my desktop) issue with MOS crashing locally when issuing `mos build`

Issue 3:
```
$ vagrant init ubuntu/yakkety64
...
$ vagrant up
...
$ vagrant ssh
...
ubuntu@ubuntu-yakkety:~$ curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/bash
Downloading https://mongoose-os.com/downloads/mos-release/linux/mos ...
######################################################################## 100.0%
Installing into /home/ubuntu/.mos/bin/mos ...
Adding /home/ubuntu/.mos/bin to your PATH in /home/ubuntu/.bashrc
SUCCESS: /home/ubuntu/.mos/bin/mos is installed.
Run '/home/ubuntu/.mos/bin/mos --help' to see all available commands.
Run '/home/ubuntu/.mos/bin/mos' without arguments to start a simplified Web UI installer.
ubuntu@ubuntu-yakkety:~$ mkdir x
ubuntu@ubuntu-yakkety:~$ cd x
ubuntu@ubuntu-yakkety:~/x$ . ~/.bashrc 
ubuntu@ubuntu-yakkety:~/x$ mos init --arch esp32
Downloading empty app...
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x8 pc=0x84ac34e]

goroutine 1 [running]:
main.initFW(0x89417e0, 0xd770a07c, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/initfw.go:39 +0x12e
main.run(0xd7834480, 0x89417e0, 0xd770a07c, 0x0, 0x0, 0x0)
	/go/src/cesanta.com/mos/main.go:124 +0x6c
main.main()
	/go/src/cesanta.com/mos/main.go:211 +0x418
```

Issue 4:
(I completely deleted `~/.mos`)

```
$ mos build
fatal error: unexpected signal during runtime execution
[signal SIGSEGV: segmentation violation code=0x1 addr=0x63 pc=0xc50d1b46]

runtime stack:
runtime.throw(0x866ffe2, 0x2a)
	/usr/local/go/src/runtime/panic.go:596 +0x7c
runtime.sigpanic()
	/usr/local/go/src/runtime/signal_unix.go:274 +0x28c

goroutine 36 [syscall, locked to thread]:
runtime.cgocall(0x84c15e0, 0xd797c6e4, 0x866f3ba)
	/usr/local/go/src/runtime/cgocall.go:131 +0xe1 fp=0xd797c6c8 sp=0xd797c6ac
net._C2func_getaddrinfo(0xc4600470, 0x0, 0xd78bd1c0, 0xd78a6758, 0x0, 0x0, 0x0)
	net/_obj/_cgo_gotypes.go:85 +0x57 fp=0xd797c6e4 sp=0xd797c6c8
net.cgoLookupIPCNAME.func2(0xc4600470, 0x0, 0xd78bd1c0, 0xd78a6758, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/cgo_unix.go:151 +0x133 fp=0xd797c710 sp=0xd797c6e4
net.cgoLookupIPCNAME(0xd78bce00, 0xf, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/usr/local/go/src/net/cgo_unix.go:151 +0x173 fp=0xd797c790 sp=0xd797c710
net.cgoIPLookup(0xd78ac900, 0xd78bce00, 0xf)
	/usr/local/go/src/net/cgo_unix.go:203 +0x2d fp=0xd797c7e0 sp=0xd797c790
runtime.goexit()
	/usr/local/go/src/runtime/asm_386.s:1629 +0x1 fp=0xd797c7e4 sp=0xd797c7e0
created by net.cgoLookupIP
[... 77 lines cut ...]
net/http.(*Transport).dial(0x8937ee0, 0x89417e0, 0xd77d00cc, 0x8650940, 0x3, 0xd78bce00, 0x13, 0xd77f2780, 0xc7583258, 0x0, ...)
	/usr/local/go/src/net/http/transport.go:849 +0x17c
net/http.(*Transport).dialConn(0x8937ee0, 0x89417e0, 0xd77d00cc, 0x0, 0xd78ac5c0, 0x5, 0xd78bce00, 0x13, 0x84b30fe, 0x8665e6e, ...)
	/usr/local/go/src/net/http/transport.go:1008 +0x174a
net/http.(*Transport).getConn.func4(0x8937ee0, 0x89417e0, 0xd77d00cc, 0xd78bce20, 0xd78ac640)
	/usr/local/go/src/net/http/transport.go:908 +0x3e
created by net/http.(*Transport).getConn
	/usr/local/go/src/net/http/transport.go:910 +0x2b0

goroutine 35 [select]:
net.cgoLookupIP(0x8941820, 0xd78ac880, 0xd78bce00, 0xf, 0x32, 0x3a, 0xd78ac5c0, 0xd78bcd00, 0x3a, 0x32)
	/usr/local/go/src/net/cgo_unix.go:214 +0x252
net.(*Resolver).lookupIP(0x8c0cff3, 0x8941820, 0xd78ac880, 0xd78bce00, 0xf, 0x0, 0x8083169, 0x0, 0xd77f173c, 0x2)
	/usr/local/go/src/net/lookup_unix.go:69 +0xc8
net.(*Resolver).(net.lookupIP)-fm(0x8941820, 0xd78ac880, 0xd78bce00, 0xf, 0xd78ac5c0, 0x3a, 0x0, 0x0, 0xd7830380)
	/usr/local/go/src/net/lookup.go:156 +0x40
net.glob..func11(0x8941820, 0xd78ac880, 0xd78a6728, 0xd78bce00, 0xf, 0x25, 0x86588a1, 0xc, 0x84b1bd3, 0xd78ac5c0)
	/usr/local/go/src/net/hook.go:19 +0x3c
net.(*Resolver).LookupIPAddr.func1(0x3a, 0x8295dae, 0x8937ee0, 0x89417e0)
	/usr/local/go/src/net/lookup.go:162 +0x45
internal/singleflight.(*Group).doCall(0x8bf6cb0, 0xd78a8ba0, 0xd78bce00, 0xf, 0xd78bcec0)
	/usr/local/go/src/internal/singleflight/singleflight.go:93 +0x21
created by internal/singleflight.(*Group).DoChan
	/usr/local/go/src/internal/singleflight/singleflight.go:86 +0x26d
```

dimonomid, 2017-07-19:
Thanks for the summary.

(2) is fixed: check out the readme https://github.com/cesanta/mongoose-os/tree/master/mos , tested in clean vagrant yakkety instance (note that go-bindata and go-bindata-assetfs are fetched as part of `make -C mos install`)

(3) is fixed in latest mos (if you use release version, please do `mos update latest`; we'll cut the release soon). For some reason, the C implementation of DNS resolver causes the issue, for now we've forced the usage of pure-go DNS resolver

Totally no idea about (4) yet.

MrSurly, 2017-07-19:
Looking good.  Closing this -- will look into MOS install on my box.

closed 2017-07-19 by MrSurly