cesanta/mongoose-os #298: MQTT in the new release (1.4)

pherig, 2017-06-21:
What happened to the MQTT files in the new release (1.4)?

I cannot build my project anymore:
/app/src/backend.c:8:30: fatal error: fw/src/mgos_mqtt.h: No such file or directory
 #include "fw/src/mgos_mqtt.h"

I checked the github and this file exist in the previous version (1.3), but is not present in the newer one (1.4)

cristiand391, 2017-06-21:
Did you include the mqtt library in mos.yml file?

pherig, 2017-06-21:
Thank you @cristiand391.

I have been developing my project for a while and I didn't notice the update that add the libs in the mos.yml file.

pherig, 2017-06-21:
If someone face this issue in the future, use the following snippets:

mos.yml:
```
libs:
- origin: https://github.com/mongoose-os-libs/mqtt
```

your code:
`#include "mqtt.h"`

closed 2017-06-21 by pherig