cesanta/mongoose-os #301: Support with esp8266 for Wiznet 5100 or ENC28J60!

tusharvb19, 2017-06-23:
I want to use Ethernet interface along with os for its features!
Need help!
TIA

tectiv3, 2017-07-04:
I started working on it:
https://github.com/tectiv3/lwip-enc28j60-app
https://github.com/tectiv3/lwip-enc28j60
It is a work in progress as currently I can't initiate connections from mongoose.

cpq, 2017-07-15:
@rojer please integrate once @tectiv3 finishes

tusharvb19, 2017-07-17:
Waiting for the Ethernet support...LAN8720 and Wiznet5100😊

cpq, 2017-07-22:
Done for esp32.

closed 2017-07-22 by cpq

tusharvb19, 2018-02-10:
what about esp8266? If I want to connect it with W5100 or other Ethernet chip?

cpq, 2018-02-10:
@tusharvb19 , your contribution of ethernet support for esp8266 would be welcome.

tusharvb19, 2018-02-10:
It uses SPI and has ready arduino ethernet library.
But how to use in mongoose os?