cesanta/mongoose-os #304: Unable to build locally on windows

wallaroo, 2017-06-27:
hi,
i'm trying to build a fresh initiated project on windows w/o success on windows:
`
mkdir mos
cd mos
mos init
mos build --local --arch esp32
Handling lib "rpc-service-config"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\rpc-service-config"
Handling lib "rpc-common"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\rpc-common"
Handling lib "http-server"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\http-server"
Handling lib "atca"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\atca"
Handling lib "rpc-service-fs"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\rpc-service-fs"
Handling lib "rpc-common"...
Already handled, skipping
Handling lib "rpc-uart"...
The --lib flag was not given for it, checking repository
Prepared local dir: "C:\\Users\\michele\\.mos\\libs\\rpc-uart"
Handling lib "rpc-common"...
Already handled, skipping
The flag --repo is not given, going to use mongoose-os repository
Sources: [C:\Users\michele\.mos\libs\http-server\src\*.c C:\Users\michele
:\Users\michele\.mos\libs\rpc-common\src\mg_rpc\*.cpp C:\Users\michele\.m
hele\.mos\libs\rpc-service-config\src\*.c C:\Users\michele\.mos\libs\rpc-
hele\.mos\libs\rpc-service-fs\src\*.cpp C:\Users\michele\.mos\libs\rpc-ua
cpp D:\mos2\build\gen\deps_init.c]
Building...
docker: Error parsing reference: "docker.cesanta.com/esp32-build:2.0-r8\r
See 'docker run --help'.
Error: exit status 125
`

Docker version 17.03.1-ce, build c6d412e

wallaroo, 2017-06-27:
in windows bash i cant install mos cli:

wallaroo@MICHELE-NB:~$ curl -fsSL https://mongoose-os.com/downloads/mos/install.sh | /bin/bash
Downloading https://mongoose-os.com/downloads/mos/linux/mos ...
######################################################################## 100,0%
Installing into /home/wallaroo/.mos/bin/mos ...
Adding /home/wallaroo/.mos/bin to your PATH in /home/wallaroo/.bashrc
SUCCESS: /home/wallaroo/.mos/bin/mos is installed.
Run '/home/wallaroo/.mos/bin/mos --help' to see all available commands.
Run '/home/wallaroo/.mos/bin/mos' without arguments to start a simplified Web UI installer.

~$ .mos/bin/mos
bash: .mos/bin/mos: cannot execute binary file: invalid executable format

dimonomid, 2017-08-18:
Local build on windows is supported now (in both "Docker Toolbox" and "Docker for Windows").

closed 2017-08-18 by dimonomid